ALIGN-analoglayout/ALIGN-public #92: Make flow not working in master

stevenmburns, 2019-07-20:
@kkunal1408 The make flow is now working currently. Can you fix it?
Pull request #88 has the circleci config necessary to include make runs in the CI.

stevenmburns, 2019-07-22:
This seems to be the first issue:
```text
if [ ! "$(docker ps -a -f name=topology_container)" ]; then docker stop topology_container; fi
if [ "$(docker ps -aq -f status=exited -f name=topology_container)" ]; then docker rm topology_container; fi
topology_container
docker run --name topology_container --mount source=inputVol,target=/INPUT topology bash -c "source /sympy/bin/activate && cd /DEMO/ && ./runme.sh telescopic_ota && cp -r ./Results /INPUT"
Start annotating DESIGN Name : telescopic_ota
Reading library file:  basic_library/basic_template.sp
INFO: PARSING LIBRARY FILE DONE
Reading library file:  basic_library/user_template.sp
INFO: PARSING LIBRARY FILE DONE
Reading Library Successful. Graphs are stored in library_graphs
Reading netlist file: telescopic_ota.sp
INFO: PARSING INPUT NETLIST FILE DONE
[]
circuit graph written in dir: circuit_graphs
Start matching graphs
OUTPUT LEF generator: ./Results/switched_capacitor_filter_lef.sh
OUTPUT verilog netlist at: ./Results/switched_capacitor_filter.v
docker cp topology_container:/INPUT/Results ./sub_circuit_identification/
```
The DESIGN_NAME should be  `telescopic_ota` but `switched_capacitor_filter` is used here.

kkunal1408, 2019-07-22:
Where can I see this error? 
It's because I had an assumption that I have only one circuit graph in the directory. I am removing other graphs from "circuit_graphs" directory from runme.sh file. It may be that remove command is not working within the circleci.

Command inside runme file--> "rm -rf ./circuit_graphs/* LOG/* ./results/*.p library_graphs/* " .

Where to check the error you have placed?
or I can pass design_name at each stage.  Design_name is an input parameter only at circuit graph generation stage, I can add it as parameter to other stages as well.
Kunal

kkunal1408, 2019-07-23:
Fixed it. It should be "Results" and not "results". Will check rest of the flow and update.

kkunal1408, 2019-07-23:
PDK abstraction is missing "v0" connection. Assertions failing for mock PDK. @parijatm @arvuce22

parijatm, 2019-07-23:
Was unable to reproduce "v0" missing error. Please provide error signature / steps to reproduce.

I did run into some issues with default python / pip versions when attempting to run make. Not sure why they cropped up for my setup alone. Please confirm that the latest commit to #88 works for your setup as well.

ALIGN-analoglayout, 2019-07-24:
Being addressed by #88

closed 2019-07-24 by ALIGN-analoglayout