ALIGN-analoglayout/ALIGN-public #94: Handle python3 dependency in a more platform-agnostic fashion

parijatm, 2019-07-25:
python3 & pip3 are currently hard-coded in Makefile. 

May be better to create a virtual environment in a separate step.

stevenmburns, 2019-08-15:
I made one step forward for this in #116
The previous shbang lines were calling python2 (`/usr/bin/python`). At least now we are using `/usr/bin/env python` which will use the python3 if you are in a virtual environment.

I propose the we always work in a python3 virtual environment so that python means python3.
They are very easy to make python3.6. Just call `python -m venv <path/to/virtual_env>` to create, `source <path/to/virtual_env>/bin/activate` to activate, and `deactivate` to deactivate.

stevenmburns, 2019-09-05:
We have done this in the docker-compose based flows.

closed 2019-09-05 by stevenmburns