ALIGN-analoglayout/ALIGN-public #101: PnR crashes if you use a capacitor and don't have a corresponding CC constraint

stevenmburns, 2019-07-28:
@Lastdayends PnR seg. faults if you don't have a CC constraint corresponding to the capacitors in the design (I guess the CC constraint causes the capacitor array to be generated---otherwise there are pins with missing components in the placer.)

We should probably explicitly check for this and assert with a decent error message. It took a while for me to figure out what was going one.

Lastdayends, 2019-07-29:
Hi Steven,

You are right. If CC constraint is missed, no common centroid capacitor
will be created, thus the pin information is empty for this common centroid
capacitor.

I will let the code print something for CC constraint missing error.

Yaguang

Lastdayends, 2019-07-29:
Hi Steven,

By the way, which design are you testing? (dc_dc?) Can you send me the link
of your test case? I can check your case about this CC format issue.

Yaguang

Yaguang

stevenmburns, 2019-10-03:
Closing because this has been superseded by #187 . Same issue, different case

closed 2019-10-03 by stevenmburns