ALIGN-analoglayout/ALIGN-public #102: Long PnR runtime for examples/sc_dc_dc_converter

stevenmburns, 2019-07-28:
@Lastdayends It seems that this testcase takes a long time to run. A lot of time is spent in the detailed router in what I think is the dijkstra routine.

stevenmburns, 2019-07-28:
Maybe we are not expecting large graphs:
```
...
Start Creating adjacent list (graph), End creating adjacent list (graph)
Detail Router check point 4
Path No 1 current path index 0
start dijkstra, 
checkpoint 0
graph.size() 2130185
check point 0.1
check point 0.2
check point 0.3
checkpoint 1
checkpoint 2
...
```

Lastdayends, 2019-07-28:
Hi Steven,

I see.  Let me check it.

Yaguang

stevenmburns, 2019-07-29:
I've made some progress on this. From more than one hour down to 12 min. or so. I'll check my changes into a branch and have you review the PR.

Lastdayends, 2019-07-29:
Hi Steven,

Sure. Let me review your change. What is the branch name？

Yaguang

stevenmburns, 2019-07-29:
Take a look at pull request #103   The branch name is feature/even_more_testcases (you can see that in the pull request.) The main thing I did was add a "break" so that at most one item is returned in one of your subroutines of dijkstra (you were only looking at one anyway.)

Lastdayends, 2019-07-29:
Hi Steven,

You can merge it to the master now.

Yaguang

Lastdayends, 2019-07-29:
Hi Steven,

I noticed that the common centroid capacitor in both v file and const file
is not correct. So I modified it.

Please check the it.

Yaguang

ALIGN-analoglayout, 2019-07-31:
I'm closing this because major issues were addressed in #103

closed 2019-07-31 by ALIGN-analoglayout