ALIGN-analoglayout/ALIGN-public #105: Very large LEF file (3 GB) generated in the sc_dc_dc_converter testcase

ALIGN-analoglayout, 2019-07-29:


Lastdayends, 2019-07-29:
> @arvuce22
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_arvuce22&d=DwMCaQ&c=u6LDEWzohnDQ01ySGnxMzg&r=ZMUCBc0sWFUe5MzbQ3jL6w&m=ifgSTNJ0VuRob7Oc2OAoVVyCbKqsJ19BFGcTgGAoM3I&s=-89W8brokh7ViT47EkWu64LhSqwUpwsI24E9F3bfsH0&e=>
> @kkunal1408
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_kkunal1408&d=DwMCaQ&c=u6LDEWzohnDQ01ySGnxMzg&r=ZMUCBc0sWFUe5MzbQ3jL6w&m=ifgSTNJ0VuRob7Oc2OAoVVyCbKqsJ19BFGcTgGAoM3I&s=r7zSmhW3OiwucdQTySXr-HG468DPlPnSk-Q28CdZXyE&e=>
> @Lastdayends
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_Lastdayends&d=DwMCaQ&c=u6LDEWzohnDQ01ySGnxMzg&r=ZMUCBc0sWFUe5MzbQ3jL6w&m=ifgSTNJ0VuRob7Oc2OAoVVyCbKqsJ19BFGcTgGAoM3I&s=8xY5DlkON9PI9CDYXebCDt2eNjjaO3r1trjA8xYB9ZM&e=>
> One bottleneck in runtime for this example is that PnR takes a long time
> to read in the LEF file. It seems to be very large and from what I can tell
> some of the component descriptions are repeated 10000 times in the file. I
> think we need only one copy of each template.
>
> (general) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/PlaceRouteHierFlow$ egrep MACRO ../testcase_latest/sc_dc_dc_converter.lef | sort | uniq -c
>     749 MACRO cap_10f
>      72 MACRO cap_12f
>    9924 MACRO CMC_NMOS_n10_X2_Y2
>    4998 MACRO CMC_NMOS_n10_X3_Y1
>     749 MACRO CMC_NMOS_n10_X40_Y25
>       7 MACRO CMC_NMOS_n12_X139_Y6
>     378 MACRO CMC_NMOS_n12_X2_Y1
>    9924 MACRO CMC_NMOS_n12_X3_Y1
>    9924 MACRO CMC_PMOS_n10_X3_Y1
>    4998 MACRO CMC_PMOS_n10_X4_Y3
>    9924 MACRO CMC_PMOS_n12_X2_Y1
>     378 MACRO CMC_PMOS_n12_X5_Y2
>    4998 MACRO CMC_PMOS_S_n10_X1_Y1
> [... 32 lines cut ...]
>    9924 MACRO Switch_PMOS_n12_X2_Y1
>     378 MACRO Switch_PMOS_n12_X5_Y1
>     378 MACRO Switch_PMOS_n12_X5_Y2
>     288 MACRO Switch_PMOS_n12_X5_Y4
>    9924 MACRO test

Hi Kunal,

Is this lef file generated by software tool?

Yaguang

ALIGN-analoglayout, 2019-07-29:
I wrote a python script to just take one copy of each of the macros we need that that brings the runtime down to 9 mins. We should still figure out how this is happening and fix it.

ALIGN-analoglayout, 2019-07-31:
It seems that something is just concatenating all the .lef files (including other previously concatenated .lef files) into a new .lef file. This can lead to exponential growth. The fix is to remove the .lef files from the source directory. THis should be fixed when we no longer generated files in source directories.

closed 2019-07-31 by ALIGN-analoglayout