ALIGN-analoglayout/ALIGN-public #108: Place and Route for telescopic ota is not on grid and visually has design rule errors

stevenmburns, 2019-08-02:
@Lastdayends @arvuce22 @kkunal1408 
I was looking through the result of the full flow on the telescopic_ota and I see a few things that we need to fix:
* When we make a connection from new metal3 down to our metal2 pins we seem to be adding off-grid metal2 (not quite creating a short but definitely a design rule violation) A good example is the left side metal3 in the picture below.
* We seem to be doing the same thing when we add metal2 pins. They don't seem to be on-grid. See (vinn, voutp,vbiasn) in the picture below.
[image]

stevenmburns, 2019-08-03:
Any thoughts on what is going on here. Do you need me to setup a reproducible testcase?

Lastdayends, 2019-08-03:
Hi Steven,

I guess the width or height of the building blocks is not integral times of
corresponding metal pitches, so shifting exists.
Yes, please give me a reproducible testcase so that I can look into it.

Yaguang

Lastdayends, 2019-08-05:
Hi Steven,

Is the reproducible testcase ready?

Yaguang

stevenmburns, 2019-08-06:
Opened pull request #112 with this data. Look at the README.md in run.

closed 2019-08-06 by stevenmburns