ALIGN-analoglayout/ALIGN-public #118: Replace the *_lef.sh primitive generation bash script with something in Python

stevenmburns, 2019-08-13:
@arvuce22 @kkunal1408 
Currently we using python to write out a bash script, that when run calls multiple python scripts.
There are multiple problems with this but the main one is that we aren't catching errors (if one of the primitive fails to generate we don't know.)

There are several ways to fix this. One idea is to just import all the python primitive generation code into the python script that current writes out the bash script and generate the primitives from there.
If we want more decoupling, we can write out a json file with the kinds for primitives we want to generate and use that as input to the primitive generation script.

parijatm, 2019-10-31:
I will take a first stab at this as part of a bigger package dependency / flow rewrite. Assigning to self.

parijatm, 2020-01-03:
Addressed by #273.

closed 2020-01-03 by parijatm