ALIGN-analoglayout/ALIGN-public #123: Can't make compose flow work in either mode

stevenmburns, 2019-08-21:
@desmonddak 

This could be a documentation issue.
The first suggested flow fails because there is no virtual environment in "/general" to source. (It is installed elsewhere and probably should be assumed to be activated.)

The second suggested flow fails because the last line is incorrect:
```
docker-compose exec make-docker-service make -f $ALIGN_HOME/DESIGN=<design>
```
Tried a bunch of other things but couldn't figure it out.

stevenmburns, 2019-08-21:
Addressed by #124

closed 2019-08-21 by stevenmburns

desmonddak, 2019-08-21:
I fixed the README:   the command needs to be
make BUILD=docker -f $ALIGN_HOME/compose/Makefile DESIGN=<design>

I created a pull request to update this documentation.

I deliberated on making the default behavior of the Makefile to execute commands via Docker containers or in the native environment.   Right now the default is native, so we need to tell it BUILD=docker to get it to invoke containers.   The make-docker-service has almost no software, so it fails to find /general.  I should probably find a way to set BUILD as an environment variable within the make-docker-service so that there is no need to specify the flag to use containers.