ALIGN-analoglayout/ALIGN-public #155: Cap array generator is producing m1 pins that are not on the routing grid

stevenmburns, 2019-09-05:
It seems that metal1 pins generated in the cap array are not on grid, so the top-level router does not properly connect to them (vias are missing). You can see this in the switch_capacitor_filter and especially in the subblock switch_capacitor_combination.

This is more our CI designs in examples/ using the Mock PDK.

stevenmburns, 2019-09-06:
@Lastdayends 
I added a complete testcase to Bugs in the feature/try_m3_pins branch #152

stevenmburns, 2019-09-06:
I've added something to #152 to help with this debug.
Follow the direction in Bugs/README.md

Lastdayends, 2019-09-06:
Hi Steven,

Thanks. I will look into it then reply to you.

Yaguang

stevenmburns, 2019-09-11:
@Lastdayends @desmonddak @parijatm @arvuce22 @sapatnekar 

I've been trying to work on this. The cap_placer code is very complex for what it seems to be doing. Should we consider moving the 9000 lines of code here into a few hundred lines of Python using  our Python-based primitive generation framework?

stevenmburns, 2019-09-11:
It seems that @Lastdayends put in a fix last night. We can delay this discussion for a while.

Lastdayends, 2019-09-11:
Hi Steve,

Actually, I think the M1 off-grid problem has been solved. It's a small bug
in my code. Please check the branch, bug/cap_M1_off_grid.

Yaguang

stevenmburns, 2019-09-11:
The fix in #166 does not resolve all the pin misalignment.
Here is an example in the same testcase run again through placement and routing.

[image]

You can examine the LEF files generated by the cap_placer to see the pin misalignment.

Lastdayends, 2019-09-11:
Hi Steven,

I see.

Yaguang

stevenmburns, 2019-09-11:
Here is what we are getting for Cap_30fF_Cap_30fF:

```
(general) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/Bugs$ head -20 Results/Cap_30fF_Cap_30fF_AspectRatio_1x6.lef
MACRO Cap_30fF_Cap_30fF
  ORIGIN 0 0 ;
  FOREIGN Cap_30fF_Cap_30fF 0 0 ;
  SIZE 10.16 BY 26.208 ;
  PIN MINUS1
    DIRECTION INOUT ;
    USE SIGNAL ;
    PORT 
      LAYER M1 ;
        RECT 3.68 25.752 3.712 25.824 ;
      LAYER M2 ;
        RECT 3.66 25.772 3.732 25.804 ;
      LAYER M1 ;
        RECT 6.976 25.752 7.008 25.824 ;
      LAYER M2 ;
        RECT 6.956 25.772 7.028 25.804 ;
      LAYER M2 ;
        RECT 3.696 25.772 6.992 25.804 ;
    END
  END MINUS1
(general) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public/Bugs$ python
Python 3.6.8 (default, Aug 20 2019, 17:12:48) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 3.68*2000
7360.0
>>> 3.68*2000 % 80
0.0
>>> 3.712*2000 % 80
64.0
>>> 6.976*2000 % 80
32.0
>>> 7.008*2000 % 80
16.0
>>> 
```

Lastdayends, 2019-09-11:
Hi Steve,

Thanks.

Yaguang

Lastdayends, 2019-09-19:
Hi Steven,

The via misalignment here is not a problem.
There are three vias, v1 (m1-m2), v2 (m2-m3) and v3 (m3-m4). v2 and v3 are
aligned, while v1 is not.
Only only restriction for via is the distance restriction between via at
same via layer. Are there other restrictions for via?

Yaguang

Lastdayends, 2019-09-19:
> @Lastdayends
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_Lastdayends&d=DwMCaQ&c=u6LDEWzohnDQ01ySGnxMzg&r=ZMUCBc0sWFUe5MzbQ3jL6w&m=nDozwZo1uCEGVU3NUFO-IbyqR-SBomlVwH3rO__qvaQ&s=aBnEqjsTS6eH69j8ERQe32KAWby2seGJP0kUDaJ8ZNw&e=>
> Is the off grid pin issue something that can be fixed easily? Let me know.
> I think the cap_placer / router is more a component generation task and
> maybe should be done using our Python infrastructure?

Here is the layout.

[image: image.png]

stevenmburns, 2019-10-03:
This was addressed in #173

closed 2019-10-03 by stevenmburns