ALIGN-analoglayout/ALIGN-public #156: Via enclosures not correctly implemented in the TAMU detailed router

stevenmburns, 2019-09-05:
@Lastdayends 
[image]
This issue still exists.
I think the Mock PDK JSON is correct.

This shows up in the PnRDB for all m3 wires with v2 connections.

The telescopic_ota examples/ shows this.

Lastdayends, 2019-09-05:
I will look into it.

Lastdayends, 2019-09-05:
Hi Steve,

The v2 via model in json file is not correct. If you set VencP_H to 0, the
problem should be fixed.

[image: image.png]

Yaguang

stevenmburns, 2019-09-05:
@arvuce22 This is not how I understand it. Should we have a short meeting with Arvind to discuss this?

Lastdayends, 2019-09-05:
@Burns, Steven M <steven.m.burns@intel.com>  @arvuce22 s

It's necessary to check with Arvind. From our understanding, the enclosure
is (metal width + 2 * VencP). While arvuce suggests (via width + 2*VencP).
That's the difference.

Yaguang

Lastdayends, 2019-09-05:
> @Burns, Steven M<mailto:steven.m.burns@intel.com>  @arvuce22 s

Yes. That makes sense.
Also, do you know what the difference is between VencP_L and VencP_H?

arvuce22, 2019-09-05:
VencP_L: This is for a lower metal layer in a stack
VencP_H: This is for an upper metal layer in a stack
e.g., For V2 stack is M2-M3, therefore, VencP_L is corresponding to M2 and
VencP_H for M3.

Lastdayends, 2019-09-05:
Hi Steven,

VencP is the metal enclosure length perpendicularly to origin metal
direction. VencP_L is the metal enclosure for lower metal, VencP_H for
higher metal.

Yaguang

stevenmburns, 2019-09-05:
Resolved by PR #159

closed 2019-09-05 by stevenmburns