ALIGN-analoglayout/ALIGN-public #161: Generate layer-specific global route information for use with Intel Analog Detailed Router

stevenmburns, 2019-09-07:
@Lastdayends 

Has the new Gcell based global router been modified so it can generate layer-specific global route information?

If not, what do you need from me to get started? Or alternatively, what do I need to know to do it myself?

Lastdayends, 2019-09-07:
Hi Steven,

Actually, integration Intel detail router includes two part:
1. P&R flow generates layer-specific global route information for Intel
detail router. (In this part, new Gcell-based global router been modified
and can generate layer-specific global route information. But needs some
coding to extract global route information to HierNode then write out write
out global route information in the format we mentioned. A new student,
Yishuang, is working on this part. We can add him in the loop.)
2. P&R flow accepts Intel detail router information to hierNode. I think we
can start work on this part, even though the part 1 is not finished yet.

The information that P&R needs after detailed router is quite simple,

For each net, the metal information and via information are needed. The
metal information includes, metal layer, metal start and end point,
metal rectangle. The via information includes, via model index, via
rectangle, via lower and upper enclosure metal rectangle.

Let me know if you have any questions.

Yaguang

stevenmburns, 2019-09-08:
Ok. I'll start on part 2) and to the rest (besides global route info) of generating information for the ADR (terminals, etc.)

stevenmburns, 2019-10-03:
I think we can close this now. An initial implementation in is master. Thank you.

closed 2019-10-03 by stevenmburns