ALIGN-analoglayout/ALIGN-public #165: TAMU Router does not appear to respect MinL constraints

parijatm, 2019-09-10:
Can you please take a look at this @Lastdayends ?

Lastdayends, 2019-09-10:
@parijatm 

We do not provide this constraint. If it's necessary, we can develop it.

Yaguang

parijatm, 2019-09-10:
Let us discuss this during our next group meeting. What do you think the effort involved will be?

Lastdayends, 2019-09-11:
Hi Parijat,

This minL constraint should specify the distance/resistance between two
pin. So the effort should be adding a objective term for this constraint in
placer. Distance or resistance, which do you think make more sense?

Yaguang

stevenmburns, 2019-09-11:
No, the minL constraint is the minimum length of a wire. There are very short wires in the layout. Every wire needs to be at least this long.

Lastdayends, 2019-09-26:
@parijatm 

What's the testcase for this error? I will run and look into the layout.

Yaguang

parijatm, 2019-09-26:
The simplest example to check out is the telescopic OTA.
'MinLength violation on M2: net012[5688, 9544, 5832, 9608]'
'MinLength violation on M3: voutn[4280, 5800, 4360, 5960]'
'MinLength violation on M3: net06[4280, 8824, 4360, 8984]'

The layout was generated a while back so let me know if you can't find the issues and I will look again.

Lastdayends, 2019-09-26:
Thanks.

Yaguang

stevenmburns, 2019-10-03:
@Lastdayends 
This is still an issue. You can see a report when you run pytest in the PnRPython directory.

Lastdayends, 2019-10-06:
@stevenmburns 

I am looking at it.

Yaguang

Lastdayends, 2019-10-06:
@parijatm @stevenmburns 

I found a fundamental logic bug for grid construction in our code, which may results minL error. But the length of M2 in this case, 72 (144/2), is strange. If you are using mock PDK, 74, should never happens. 80, or multiple 80 is possible (because M1/M3 pitches are 80). Which PDK are you using?

Yaguang

parijatm, 2019-10-06:
Sounds like you have enough metal being generated to encapsulate a single via (via_width = 32 & via_enc = 20) results in a metal length of 32 + 20 + 20 = 72. Please check the layout to make sure. 

Are you generating spurious metal rectangles everytime you generate a via? I remember seeing something to that effect in a very early version of the generated GDS. If that's the case, you are probably observing these errors at the primitive pin connections.

Lastdayends, 2019-10-06:
@parijatm 

Yes. Those metals are via enclosure metals.  Those enclosure metal is created for each via, so this is always MinL error. 

Should we just ignore this issue? Or do you have any suggestion for the writing of enclosure metal.

Yaguang

Lastdayends, 2019-10-07:
Hi Parijat,

I can somehow modify codes to aviod minL issue due to via enclosure metal.
But I remember in other software (like cadence), separately writing via
enclosure metal should always not lead to minL issue, even if via enclosure
metal is smaller than minL. What's your suggestion?

Yaguang

stevenmburns, 2019-10-07:
I think this depends on process. In Intel processes, via enclosures are not always enough to satisfy min length, so antennas need to be added, on one or both sides. This is a critical part of the detailed router.
However, I don’t know if that is true for GF 12. @arvuce22 Do enclosure metals on intermediate vias (stacked vias) need to satisfy min length rules in GF 12? If not, how should we specify what then do need to satisfy?

arvuce22, 2019-10-07:
Yes, It is true for GF also.

parijatm, 2019-10-07:
> Hi Parijat, I can somehow modify codes to aviod minL issue due to via enclosure metal. But I remember in other software (like cadence), separately writing via enclosure metal should always not lead to minL issue, even if via enclosure metal is smaller than minL. What's your suggestion? Yaguang

Separately writing via enclosure metal should not lead to minL issue **IF** it overlaps with a metal rectangle on the same layer that does not violate minL. The issue in your case is being caused by pins of the type shown below:

[image]

These do not connect down to any primitive so the MinL is a valid issue. There is a chance that these MinL issues will go away when we connect this block at a higher level but there is no real guarantee. 

As of #191, we do not see these same issues on the primitive connections anymore since this type of connection overlaps with a MinL satisfying M2 rectangle within the primitive.

parijatm, 2019-10-07:
In terms of solutions.... Why are you drawing the M2 shapes to begin with?

Lastdayends, 2019-10-07:
@Burns, Steven M <steven.m.burns@intel.com>  @Mukherjee, Parijat
<parijat.mukherjee@intel.com>

I see.

Yaguang

Lastdayends, 2019-10-10:
@parijatm 

I think this problem is due to the difference between ASAP7 and Mock PDK.

In ASPA7, the minL of metal wire is the metal pitches, so the via metal enclosure will not lead to a drc problem. But in Mock PDK, minL is much bigger than the metal pitches, so for the terminal, there will always be a problem if the code follows the rule of ASAP7.

One way to solve this problem, is create a metal for terminal with minL metal, so that the via enclosure metal will not be a problem.

Other way is to extent the via enclosure metal, when the via is connected to terminal. Or extent the via enclosure metal when there is a single via enclosure here.

Which one do you think is better?

Yaguang

parijatm, 2019-10-10:
Your first proposal (create a metal for terminal with minL metal) works but is overly conservative... You may end up drawing much longer metals than you really need which in turn increases the total capacitance - something many of our circuits are sensitive to. I am assuming you have other code ensuring you won't accidentally create shorts of course.

The other heuristics you described work well to get rid of the current set of bugs. But may not work under every circumstance. We may easily run into a process technology where the MinL is greater than the distance between two adjoining vias & their enclosures for example. The single via assumption and simple terminal assumption both fail in this hypothetical situation.

The best thing to do would be to have one centralized method responsible for drawing the metal rectangles and implement a MinL check (and extension) routine there.  How much refactoring would be involved to do this?

@stevenmburns & @desmonddak may have additional ideas.

Lastdayends, 2019-10-10:
@Mukherjee, Parijat <parijat.mukherjee@intel.com>

I am not sure how much refactoring would be involved to do a centralized
method responsible for drawing the metal rectangles and implement a MinL
check. Currently, drawing metal rectangles are done separately when it's
necessary. Function for MinL check is also absent.

An rough estimation about the workload to refactor this mainly includes
three part:
1. Physical pin assignment after placement (in global router stage).
2. detail router minL check function.
3. Extension function.

For the 1st item, I am not sure if it's necessary to assign physical
information for terminal. Maybe keep ignoring terminals, but that can be a
problem when we are tried to expose M2 metal to represent the terminal
(because there is no terminal at all).

Yaguang

Lastdayends, 2019-10-18:
@Mukherjee, Parijat <parijat.mukherjee@intel.com>

Some codes to check, extend metal wire as well as extendable check in
detail router, are updated. Here is an example of OTA. The wires that break
the minL rule are extended, especially for the terminals on the boundary.

[image: image.png]

If any problems, please let me know.

Thanks.

Yaguang

parijatm, 2019-10-31:
Addressed by #213

closed 2019-10-31 by parijatm

parijatm, 2019-10-31:
I was just closing the issue as resolved @Lastdayends. #213 was the pull request number where the fix was committed. 

Thank you for taking care of this !

Lastdayends, 2019-10-31:
Ok. Got that.

Yaguang