ALIGN-analoglayout/ALIGN-public #167: Fix primitive to global router pin porting

parijatm, 2019-09-13:
Original comment from Steve below:
> Let me know how you want to do the M3 porting. I currently get opens when I port both M2 and M3 lines (because I can't port the V2 without the global router crashing.)
> 
> We should do what makes sense and fix the issues that it causes.
> Options seem to be:
> 
> * [ ]  Port metal2 only --- router doesn't know that it is connected internally
> * [ ]  Port metal3 only --- router can't use existing metal2 for routes (metal2 in LEF as an obstruction)
> * [ ]  Port metal2 and metal3 --- router thinks there is an open when only looking at the LEF view (can be solved by looking at all the layout
> * [ ]  Port metal2, metal3 and via2 --- router crashes but could be fixed

parijatm, 2019-09-13:
Consensus (as discussed in #153) is that we need to eventually port metal2, metal3 & via2... Current implementation summarized by @stevenmburns below:

> I did the M3 porting a little differently in PR #152
> I ported didn't port the M2 if the M3 is ported.
> That currently helps PnR do a better job.

parijatm, 2019-11-05:
Should have been closed. Can you confirm @stevenmburns

arvuce22, 2021-04-28:
@parijatm @stevenmburns We have resolved this issue. Please reopen it if you think we have to modify other things related to the issue.

closed 2021-04-28 by arvuce22