ALIGN-analoglayout/ALIGN-public #174: DRC MinLength violations for primitives with x_cells = 1 & y_cells = 1

parijatm, 2019-09-23:
Running into DRC MinLength issues when x_cells = 1 & y_cells = 1

[image]

Potential solutions include:
- [ ] Port M1 instead of M2 when M2 is not needed for routing 
- [ ] Always port M2 or above and extend M2 rectangles where needed.

Potentially related to #167 if we decide to go with option 1.

Any feedback @stevenmburns, @arvuce22 ?

stevenmburns, 2019-09-23:
What are the grey rectangles without m2 on it.
Yes, we need to extend the m2 wire if we want to port it.
Or we can solve this issue during a fill stage (leave it as an error at this stage). It will go away later when routes connect to this m2.

[image]<https://user-images.githubusercontent.com/48996821/65469080-f73d0c80-de1a-11e9-8a8d-9d096c44080e.png>

parijatm, 2019-09-23:
> What are the grey rectangles without m2 on it. Yes, we need to extend the m2 wire if we want to port it. Or we can solve this issue during a fill stage (leave it as an error at this stage). It will go away later when routes connect to this m2. 

Gray rectangles are v0. Sorry should have hidden those.

I believe going up one layer (M2 to M3 via V2) is also a valid route for the current router. If that's the case, we can't just rely on external connections resolving this & do need some kind of fill stage.

What about the 3rd option? Does it make sense to port M1? Or does the router only operate M2 and above?

Leaving this issue open to revisit at a later point in time. Not sure we have any designs using x_cells = 1 & y_cells = 1 at the current point in time.

parijatm, 2019-10-03:
Already addressed by #184. Ended up extending rectangles where needed. We may eventually want to trim these rectangles after placement / routing to reduce wire capacitance.

closed 2019-10-03 by parijatm