ALIGN-analoglayout/ALIGN-public #187: seg-fault in Global Router

desmonddak, 2019-09-30:
Parijat has created some new leaf cells with M3 terminals and the global router seg-faults.

The bug is in Bugs/telescopic_ota with a RUNME.   The router should not fault but should report an error (in this case, I think it is a net that has no source)?

I supplied the inputs/ directory so you should be able to run pnr_compiler on these inputs and reproduce the bug.

Lastdayends, 2019-10-01:
@desmonddak 
I will lookinto it then reply to you.

Lastdayends, 2019-10-01:
@desmonddak 

What's the name of this branch? I switched some branches, but only find that Bugs/switch_capacitor_file and can not find a branch with Bugs/telescopic_ota.

Yaguang

desmonddak, 2019-10-01:
Somehow I lost track of the branch I had put it in .  It is now in branch
bugfix/multivia_primitives

The files for debug are in Bugs/telescopic_ota.   

The entire flow fails using this branch, but the core issue can be tested by running the pnr_compiler on the example.   Another way to execute it is to cd to telescopic_ota/pnr_output in that Bugs area and run make -f /PlaceRouteHier/Makefile.run DESIGN=telescopic_ota

Lastdayends, 2019-10-01:
Got it. I will look into it.

Yaguang

Lastdayends, 2019-10-01:
Hi Desmond,

The problem exists in lef file. Physical information of most pins are
missing. The global router can not find any pin information.
[image: image.png]

Yaguang

desmonddak, 2019-10-02:
Please don't seg-fault on bad or missing input.    We cannot debug flows that are being developed if each stage of the flow can only accept perfect input, or just seg-faults.  Instead, stop and report an error message, or skip then net so we can get more debugging progress.  Robustness to bad inputs is a sign of program quality.  Imagine if your C compiler crashed because an obscure library was missing but didn't tell you what was missing.  Technically the library should have been there so why would the C compiler need to check?

Lastdayends, 2019-10-03:
Hi Desmond,

I will add some warning when the physical information is missed in lef file.

Yaguang

parijatm, 2019-10-03:
Thank  you Yaguang. The missing pin issue was already fixed and committed back to master but it is important to do input validation anyway. If you can think of any other user / primitive inputs that you aren't doing some type of sanity check for, please think of adding warnings for those as well.

Lastdayends, 2019-10-03:
Thanks for your suggestions. I will add warning when reading lef file.

Yaguang

Lastdayends, 2019-10-06:
@desmonddak  

Now I add a assert when the physical pin information is missed. Please check.

Yaguang

parijatm, 2019-10-31:
Assertion added by #181. Marking this as closed.

closed 2019-10-31 by parijatm