ALIGN-analoglayout/ALIGN-public #190: All examples not in CI fail to run through the flow

stevenmburns, 2019-10-03:
@meghna09 @kkunal1408 @tonmoydhar @sapatnekar 

The following designs in $ALIGN_HOME/examples fail to run through the flow.
- [ ] sc_dc_dc_converter
- [ ] adder
- [ ] linear_equalizer
- [ ] single_to_differential_converter
- [ ] variable_gain_amplifier

The last four fail due to missing parameters.
The first fails with a core dump in the router.

We should fix these or move them somewhere else. We shouldn't be advertising test cases that fail.

stevenmburns, 2019-10-03:
@desmonddak @parijatm 
Wanted to tag you as well.

tonmoydhar, 2019-10-03:
- The designs- adder and single_to_differential_converter were working with the full flow  (with the version of ALIGN that was available on 6th September) that I installed in my local machine. We will check why these designs are failing to run. 

- Designs that involve current mirror banks, e.g., variable_gain_amplifier and linear_equalizer, were facing issues in mapping current mirror banks while running through the full-flow. We are also looking into it.  

We will update you on these.

kkunal1408, 2019-10-04:
Hi Steve,
1. sc_dc_dc_converter --> wrong constraints
2. adder/single_to_differential_converter/variable_gain_amplifier -> circuit needs to be run flat, in hierarchical implementation it gets a floating port as for transistor we are not considering body connection. 
I have pushed a branch [floating_nets](https://github.com/ALIGN-analoglayout/ALIGN-public/tree/floating_nets) for these.

Linear equalizer circuit I am working on with Arvind to fix parameters for current mirror bank. I will update once it's done.

parijatm, 2019-11-05:
Everything is working. Needs to be put in circleci @stevenmburns

kkunal1408, 2019-11-11:
For linear equalizer please use command "make DESIGN=linear_equalizer FLAT=1"

stevenmburns, 2019-11-13:
As far as I can see, only the sc_dc_dc_converter is working.
Most are giving problems in write_verilog.py

kkunal1408, 2019-11-13:
wrte_verilog_lef.py is erroring out due to 
" --width " + str(values['w']) +
" --length " + str(values['l']))" 
These commands were added for extractor.
@parijatm : should we force designer to provide w/l parameters or should I put a check on that availability of that parameter?

Removing those lines. 
PnR runs when I run as "FLAT=1" design but fails for hierarchical design.
This design was running as hierarchical in older PnR setup and some new updates in PnR is causing the issue. @Lastdayends  Can you please check the "adder" design? (Run the flow after removing the two highlighted lines)

desmonddak, 2019-11-13:
I just added the same bug on variable_gain:  Running the basic commands of read_netlist, match, write_verilog fail on exactly this width problem.    You said you are removing -- do you have a branch I can test from.  I'm trying to make sure the compose/Make works on these examples.

kkunal1408, 2019-11-13:
I don’t have a seperate branch. After removing those lines I am worried
extractor might have an issue. I will push a seperate branch by adding
those components in the netlist. I will update on the other issue once done.
Thanks
Kunal

Lastdayends, 2019-11-13:
Which branch to test it?

Yaguang

parijatm, 2019-11-13:
> @parijatm : should we force designer to provide w/l parameters or should I put a check on that availability of that parameter?

Think we should just pass whatever params we do have. I added a commit (60f722e) to #245 which addresses this. Still waiting on the CI but we can merge it if someone approves it.

kkunal1408, 2019-11-14:
Looks fine with the updates.  @Lastdayends  Can you try the #245 branch for hierarchical run? 
The run goes through with "make DESIGN=adder FLAT=1"  while gives segmentation fault with hierarchical run "make DESIGN=adder": 
make[1]: *** [Results/adder_0.lef] Segmentation fault (core dumped)

Lastdayends, 2019-11-14:
Hi Kunal,

It works for me when I run "make DESIGN=adder". What's the error
information when you run this command?

[image: image.png]

Yaguang

kkunal1408, 2019-11-14:
For me, it says: bug error 2
@desmonddak, did the flow run for other circuits for you after #245 ?

parijatm, 2019-11-14:
#245 has been merged to master.

Lastdayends, 2019-11-14:
Hi Kunal,

I know that bug. Please follow what I do as shown in the figure blow. (This
bug is due to different platform. I met this problem before but not now.)

1. Comment the code in line 1786 of capplacer.cpp in cap_placer folder.
2. Uncomment the code in line 1787.
3. Recompile the P&R, then rerun.

[image: image.png]

Yaguang

parijatm, 2019-12-26:
@kkunal1408 @Lastdayends Can this issue be closed ? Does anything still need to be added to CI?

kkunal1408, 2021-04-27:
CI is passing now for these circuits. Stale issues.

closed 2021-04-27 by kkunal1408