ALIGN-analoglayout/ALIGN-public #197: Dangerous coding style led to difficult memory bug

desmonddak, 2019-10-06:
ReadLEF.cpp: ReadLEF
          interMetals.resize(interMetals.size()+1);
          interMetals.back().metal=temp[1];

This is dangerous programming style.  It requires that you resize correctly.  In one later case,  parsing 'LAYER', the resize was entirely forgotten.    back() referred to a bad memory location.   This resulted in a seg-fault but only when not running gdb, so very difficult to find.   We should not see memory errors when using C++ properly.

The right way to code this is to create a temporary structure and then append it, which eliminates any chance of an allocation error.
       PnRDB::contact tmpCon;
       tmpCon.metal = temp[1];
       interMetals.push_back(tmpCon);

This bug was found using valgrind which showed a bad write at line 91 where a back() call was made.  A seg-fault happened only because the back() call happened on a zero length -- otherwise only bad behavior would be observed and valgrind would not have detected (we would have overwritten another contact).

We should probably eliminate any use of back() used on the LHS.
   ``

desmonddak, 2019-10-06:
Mistaken -- the vector was not empty, but the coding style is dangerous.  Still locating memory error -- it is related but not a zero vector.

desmonddak, 2019-10-06:
PnRDB/ReadLEF.cpp:85    The continue was above the resize which caused a memory issue that sometimes resulted in a seg-fault.

 if((found=def.find("LAYER"))!=string::npos) {
          temp=get_true_word(found,def,0,';',p);
          interMetals.resize(interMetals.size()+1);
          interMetals.back().metal=temp[1];
          if(temp[1].front()!='M') {continue;} // work around for obs on Via layer - wbxu 20190707

parijatm, 2019-10-31:
@desmonddak @stevenmburns Has this issue been fully resolved?

Lastdayends, 2019-10-31:
It's temporarily solved by adding an assert.

closed 2019-11-05 by parijatm