ALIGN-analoglayout/ALIGN-public #200: Excite MockPDK primitive generation code or remove untested code

parijatm, 2019-10-07:
I am not quite sure where the following primitive generators are being used:
1) fabric_CMB3_NMOS.py
2) fabric_CMB5_NMOS.py
3) fabric_Res.py

We should either be testing these modules or removing them from the codebase. 

Feels like fabric_CMB* needs to be collapsed into a more generic primitive generator & fabric_Res needs to be tested.

@arvuce22 Can you please look into this? At the very least, every MockPDK primitive generator should have simple tests that perform LVS & DRC (See FinFET14nm_Mock_PDK/test_all.py::test_ALL)

parijatm, 2019-12-26:
#241 removed a number of spurious CellFabric tests. Did we add tests / remove code from primitives too @arvuce22?

arvuce22, 2019-12-26:
@parijatm I removed some of the old primitive tests from CellFabric; we
already have these tests in the  "PDK_Abstraction/FinFET14nm_Mock_PDK/".

arvuce22, 2021-04-28:
Removed unused primitives in #667 and we are testing fabric_res

closed 2021-04-28 by arvuce22