ALIGN-analoglayout/ALIGN-public #203: Need additional PDKs in public repo

parijatm, 2019-10-08:
We should try & get as many dissimilar PDKs as we can. For example, an older bulk technology node and a dissimilar FinFET node. Intentionally creating PDKs to excite corner cases such as misaligned M1 / M3 layers is also an option.

rbarzic, 2020-01-29:
I would likely help adding a free pdk like the nangate freepdk45 but I would like to know the status of the pdks/Bulk65nm_Mock_PDK first. When I used it on some simple examples of my own, I got strange layouts that look more like a FinFet-based than a  Bulk-based layout

parijatm, 2020-02-01:
The Mock Bulk PDK is based on the Mock FinFET PDK @rbarzic so its not surprising that they look similar. As a general methodology, the generation of shapes will probably remain grid based but we will add postprocessors to modify the width to the required values. For now, the schematic has been designed such that the width is always an integral multiple of a certain width. Note that the connection to align.compiler (the part that figures out what topologies to draw and what sizes) may need additional work to fully support bulk as well (@arvuce22, @kkunal1408 any updates?)

You can definitely take a stab at adding the freepdk45 as well. Be aware there are some major changes coming to the existing primitives (See discussion in #353) so you may want to hold off for a little bit.

parijatm, 2021-11-18:
Superseded by #874

closed 2021-11-18 by parijatm