ALIGN-analoglayout/ALIGN-public #218: Routing over unit capacitors

stevenmburns, 2019-10-23:
@soneryaldiz @Lastdayends @meghna09 @desmonddak @parijatm @arvuce22 @kkunal1408 

Here is the case I was discussing yesterday in the team meeting. We are routing signals in metal5 over the top of the unit capacitors. Is this okay, or should we block off unit caps to force routing in the channels between the unit capacitors.

Let's discuss this using this GItHub issue. (If you can't see the picture below, view this issue using the GitHub web interface.)

[image]

Lastdayends, 2019-10-23:
Also the dummy flag and dummy gnd connection.

1 dummy flag in constraint to indicate whether dummy cap is needed. I can
update the constraint and coding for this part.
3 dummy gnd connection need power net in upper hierarchy to be connected. I
can update coding for this part. But need UMN team to add power net in
verillog file.

Yaguang

stevenmburns, 2019-10-23:
@kkunal1408 
 I think we want in the verilog file something like:
```
Cap_60fF c0 ( .MINUS(net63), .PLUS(net72), .MINUS_DUMMY( vss), .PLUS_DUMMY( vss)); 
```
and
```
Cap_30fF_Cap_60fF c1_c3 ( .MINUS1(net67), .P vss)LUS1(net63), .MINUS2(Voutn), .PLUS2(Vin_ota), .MINUS_DUMMY( vss), .PLUS_DUMMY( vss) );
```
Currently, there is no reference to the DUMMY terminals.

I think we want to reference both dummy terminals because that makes them like the other capacitors in the array, and we can use the existing flow to wire them together.

Lastdayends, 2019-10-23:
@Burns, Steven M <steven.m.burns@intel.com>  @Kishor Kunal
<kunal001@umn.edu>

Should vss be global gnd or gnd inner the switched capacitor combination?

If global gnd, then .minus_dummy(vss) and .plus_dummy(vss) are not
necessary. By default, any dummy pin will be connected to global gnd (power
routing). In this case, global power net should be provided in verilog file.

If gnd inner the switched capacitor combination, to keep consistent, then
when generating the verilog file, the .minus_dummy pin should be in verilog
file and the connection relationship should be indicated with the
.minus_dummy as .minus_dummy(vss). Also, since we are going to have a
dummy_flag to control whether dummy cap is needed or not. The dummy_flag
should also keep consistent with .minus_dummy in verilog file.

Yaguang

parijatm, 2019-11-05:
Dummy capacitors should be connected to analog gnd. 

No consensus yet on whether to route over or between capacitors. To discuss on Friday.

parijatm, 2021-11-11:
@arvuce22 to tag the relevant PR & close this.

arvuce22, 2022-01-20:
Resolved in #830

closed 2022-01-20 by arvuce22