ALIGN-analoglayout/ALIGN-public #222: Unclear model maps + redefined models in sub_circuit_identification

parijatm, 2019-10-25:
There are a few separate but related issues here:
1. There is no mapping from the Transistor (M*) device types represented in basic_element.sp to default devices in SPICE. We should either refactor basic_elements.sp to refer exclusively to NMOS & PMOS for M* devices... Or we should add appropriate .model statements so that the mapping is clear. I elected to take the second route in [Circuit/tests/basic_template.sp](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/131605a44bf0f3dc147f3e5450aa46527a36ca20/Circuit/tests/basic_template.sp#L1-L6) as shown below.

```
.model NMOS_VTL NMOS
.model PMOS_VTL PMOS
.model nmos_rvt NMOS
.model pmos_rvt PMOS
.model nch NMOS l=60N m=1 w=1u
.model pch NMOS l=60N m=1 w=1u
```

2. We should not be overriding default SPICE types as it results in significant confusion. For example, we are overriding Res & Cap with subcircuit definitions in basic_element.sp. This makes it very confusing as to whether future Res / Cap instantiations should start with X* or R* / C*. For the new Spice Parser in Circuit, I am simply throwing an assertion if the user attempts to redefine something that's already in the library. I have commented out the lines of concern in [Circuit/tests/basic_template.sp](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/131605a44bf0f3dc147f3e5450aa46527a36ca20/Circuit/tests/basic_template.sp#L65-L96).

```
** You may not redeclare an existing library element
; .subckt Cap PLUS MINUS
; CC1 PLUS MINUS 60f
; .ends Cap
```

```
** You may not redeclare an existing library element
; .subckt Res PLUS MINUS
; RR1 PLUS MINUS 10k
; .ends Res
```

3. Number of terminals is often ambiguous. We cannot be using 2 terminal devices to be creating 3 terminal ones. I have circumvented the issue by creating empty subcircuit definitions in [Circuit/tests/basic_template.sp](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/131605a44bf0f3dc147f3e5450aa46527a36ca20/Circuit/tests/basic_template.sp#L65-L96) as shown below.

```
** CANNOT USE 2 TERMINAL CAP TO CREATE 3 TERMINAL CAP
; .subckt Cap_b PLUS MINUS BULK
; CC1 PLUS MINUS BULK cap cap=60f
; .ends Cap_b
.subckt Cap_b PLUS MINUS BULK
.param cap=60f
.ends Cap_b
```

```
.subckt spiral_ind PLUS MINUS BULK CTAP
.param w=9u
** DEFAULT INDUCTOR IS 2 TERMINAL NOT 4.
** SUBCKT SHOULD STAND ON ITS OWN
; L0 PLUS MINUS BULK CTAP spiral_sym_ct_mu_z w=9u
.ends spiral_ind
```

**Summary:**
I have made proposed fixes in basic_template.sp in [Circuit/tests/basic_template.sp](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/131605a44bf0f3dc147f3e5450aa46527a36ca20/Circuit/tests/basic_template.sp) but this is not yet being used by sub_circuit_identification. Please let me know if there is a good reason for the redefinitions / lack of .model mapping from a system-level perspective. I believe it was mainly to construct a list of supported primitives but we should be able to create default spice devices anyway as I highlighted in #221.

kkunal1408, 2019-10-25:
The initial idea behind the redefinition of primitives (nmos/cap/res) in basic_template.sp is to make the netlist annotation technology independent. Each technology or PDK library has its own way of defining these primitives, E.g.,
For example, multiple ways of the declaration of a current mirror.
Declaration type 1:
```bash
M0 (DA DA S 0) NMOS_VTL w=w l=90n
M1 (DB DA S 0) NMOS_VTL w=w l=90n
```bash
Declaration type 2:
```bash
xM0 DA DA S 0 nmos_rvt w=w l=90n 
xM1 DB DA S 0 nmos_rvt w=w l=90n 
```bash

The cells generated from cell generator are right now not the same as NMOS_VTL/nmos_rvt so a different name was assigned for them "Switch_NMOS" using this subckt mapping (similar with cap/res). 
```bash
.subckt Switch_NMOS  D G S
M0 (D G S 0) NMOS_VTL w=w1 l=l1
.ends Switch_NMOS
```bash
Now both current mirror declaration types, as shown above, will be considered as a combination of two "Switch_NMOS" instead of "nmos_rvt/NMOS_VTL" whose device characteristics we don't know.
Here basic_template.sp is acting as a way to map any library to ALIGN. 

The other option is to get these cell names from a cell library which we don't have access to now.

parijatm, 2019-10-25:
Think I am missing something here. 

NMOS_VTL & nmos_rvt are both technology specific models of type "NMOS". So if we want to make basic_template.sp technology independent, we should be really talking in terms of the model type & not the technology-specific model (which our tool doesn't know anything about). 

I think we do need a concept of an inbuilt cell-library (which should correspond to inbuilt spice types in Xyce) and define basic_template.sp in terms of that cell-library. So the current mirror declaration in basic_template.sp should look like:

```
M0 DA DA S 0 NMOS
M1 DB DA S 0 NMOS
```

Note that I even left out w & l values since basic_template.sp is really all about topologies to the best of my understanding.

Then when we want to deal with technology specific netlists... We can just add .MODEL statements (or .SUBCKT statements) that clarify what a given model corresponds to. So basically, let's suppose we have a technology specific current mirror as part of a bigger netlist... We need to add a .MODEL line to our netlist as follows:

```
.MODEL NMOS_VTL NMOS

M0 DA DA S 0 NMOS_VTL w=w l=90n 
M1 DB DA S 0 NMOS_VTL w=w l=90n 
```

I am assuming we want the topology to be identified across all technologies. Is there some technology-specificity in subcircuit identification that I am missing or does this look like it might work? 

By the way... The .model example was for Declaration type 1. Declaration type 2 looks like nmos_rvt is a subcircuit not a transistor (instance name starts with X not M). Without knowing anything about the actual PDK, I would capture it as follows

```
.SUBCKT nmos_rvt D G S B
.PARAM w=180n l=90n
M0 D G S B NMOS w=w l=l
.ENDS
```

kkunal1408, 2019-10-25:
Yes, we can make an inbuilt cell-library like xyce as it can provide me standard. The subcircuits within basic template right now don't adhere to particular device models. My first thought of keeping it like this was if the user wants to add a new subckt inside basic_template/user_defined_template he can just copy-paste his ckt elements from his spice file. I don't care if they are named as NMOS/NMOS_VTL/NMOS_RVT/nmos_rvt, similarly values are also just placeholders in basic_template. Right now I am storing a map between device type defined in netlist and the mapped device type by parser and overite all values to the mapped device type.

But if the user wants to use the primitive layout generator the subcircuit name should match the available cell generator names. So .SUBCKT is used for mapping primitives to our library similar to MODEL line (we can switch to .MODEL type declaration for mapping res/cap/transistor). We can say I am using this basic_template for topology as well as mapping to cell generator. I think we can separate them as it provides more clarity.

I used a two-terminal device to map three-terminal device because the number of terminals in a transistor I have seen varying from 3-6. 
```bash
3 terminal when it doesn't have a separate body terminal.
5 when it has an extra deep well connection
```
similarly caps can be 2-4 terminal devices. As a workaround, I map both of them to a two-terminal device and ask designers to connect rest terminals. Now when we are moving towards LVS clean we need to standardize these.

XM0 is also an NMOS declaration variation that I see in USC circuits so I use a mixed (use the first letter and device type) kind of parser to map these.

parijatm, 2019-10-25:
> Right now I am storing a map between device type defined in netlist and the mapped device type by parser and overite all values to the mapped device type.

Storing as in manually specifying it within the code? Perhaps by doing a substring match? Feels like something the user should be doing explicitly.

And yes. Let's have a call for the number of pin issues. I am not quite sure how to tackle those as well. Alternatively, we could discuss some of this during our regular Tuesday meeting.

kkunal1408, 2019-10-25:
For each device in a netlist, I have a dictionary element that stores the mapped inst type (key "inst_type")  and device type in netlist (key "real_inst_type"). I create a bipartite graph of these dictionary elements. Tuesday is fine or we can discuss it on Monday before 3 Pm pacific time if it is blocking your work.

parijatm, 2019-10-25:
We can talk anytime between 1pm and 3pm. 

I also provided some feedback on your active pull request that references the .model stuff at least. Getting that pull request merged quickly is high priority for me. The user input to sub_circuit_identification can come later.

parijatm, 2019-10-31:
Summarizing outcomes from our discussion below:
1) There should be a list of built-in layout templates corresponding to Xyce internal types
2) We will use .MODEL statements to map derived types to their base types
3) We have observed number of capacitor pins etc vary based on technology. Need additional investigation to see how we may best represent this.

In the short term, we have modified basic_templates.sp to use Xyce internal device types (NMOS, PMOS) and the parasitic extractor is using those exact same names. Better handling of model types is required in the long run so we are keeping this issue open.

kkunal1408, 2022-01-17:
We have merged the initial part of using specific models file. 
For adding more terminal primitives we can modify the base models for specific PDKs.

closed 2022-01-17 by kkunal1408