ALIGN-analoglayout/ALIGN-public #238: sub_circuit_identification doesn't take parameters into account

parijatm, 2019-11-06:
Matching needs to take W/L ratios at least into account.

kkunal1408, 2019-11-22:
Which matching?  Do we have any specific examples?

parijatm, 2019-11-22:
Sorry I didnt go into this fully and try to come up with counterexamples yet. Just wanted to file it so that I didnt forget. 

Basically, looking through the code seems to indicate merged_value() throws away everything but the maximum value for a given parameter in merge_nodes.py. so this means that there is no way to differentiate between a 5:5 current mirror from a 1:5 current mirror when we get to write_verilog_lef. (write_verilog_lef will divide nfin with the unit cell height to arrive at the ratio). Similarly, the other primitives will create a 1:1 ratio even if the actual ratio is different.

There is a chance that this complexity is being handled elsewhere and it's only the test in test_parser that's broken. But I did try changing the nfin for both current mirror transistors in the ota.sp in src to 50 and it still generates a 1:5 current mirror. Similarly, I tried changing one of the nfins on the DP_NMOS to 25 and it still tries to generate a 1:1 dp_nmos though the real ratio is 1:3.

kkunal1408, 2019-11-22:
Yes, That was the way it's been done today as we didn't have a method to pass multiple values for the primitive generator so a max was chosen instead of providing both. I think now is the right time to move to a different way of providing inputs for passing parameters for the generator. I think in our last discussion with you we had discussed this and decided on a JSON format to provide sizing and connectivity instead of one number. Can I propose a JSON format for this annotation-cell generation integration or do you have something already?

parijatm, 2021-11-18:
Fixed by #870

closed 2021-11-18 by parijatm