ALIGN-analoglayout/ALIGN-public #247: Bug/circuit gen for variable_gain_amplifier

desmonddak, 2019-11-13:
The basic steps of read_netlist, match_graph, and write_verilog FAIL on variable_gain_amplifier:
It does not matter whether flat is set to 0 or 1.

(general) root@desmond-v01:/dataVolume/variable_gain_amplifier/topology_output#
(general) root@desmond-v01:/dataVolume/variable_gain_amplifier/topology_output# !211
python3 /ALIGN-public/sub_circuit_identification/src/read_netlist.py --dir ./input_circuit/ -f variable_gain_amplifier.sp --subckt variable_gain_amplifier --flat 1
Reading netlist file: variable_gain_amplifier.sp
INFO: PARSING INPUT NETLIST FILE DONE
circuit graph written in dir: circuit_graphs
(general) root@desmond-v01:/dataVolume/variable_gain_amplifier/topology_output# !212
python3 /ALIGN-public/sub_circuit_identification/src/match_graph.py
Start matching graphs
(general) root@desmond-v01:/dataVolume/variable_gain_amplifier/topology_output# !214
python3 /ALIGN-public/sub_circuit_identification/src/write_verilog_lef.py -U_cap 12 -U_mos 12
Traceback (most recent call last):
  File "/ALIGN-public/sub_circuit_identification/src/write_verilog_lef.py", line 422, in <module>
    ALL_LEF, UNIT_SIZE_MOS, UNIT_SIZE_CAP)
  File "/ALIGN-public/sub_circuit_identification/src/write_verilog_lef.py", line 328, in generate_lef
    " --length " + str(values['l']))
KeyError: 'w'

kkunal1408, 2019-11-14:
fixed in #245

kkunal1408, 2019-11-18:
@desmonddak It has been fixed and merged to master, did you get a chance to try it again or are you facing any other issues?

closed 2020-02-15 by kkunal1408