ALIGN-analoglayout/ALIGN-public #268: Identify cross-coupled NMOS devices

soneryaldiz, 2019-12-20:
@kkunal1408 , below is the latest identification output:
[image]

Line # 17 and 18: mn3 and mn4 is a cross-coupled pair yet it is not recognized so. CCP_* primitives require a shared diffusion. Why is this restriction needed? This is needed as these devices need matching (and possibly placed in common centroid style).

_Originally posted by @soneryaldiz in https://github.com/ALIGN-analoglayout/ALIGN-public/pull/255#issuecomment-566754350_

kkunal1408, 2019-12-20:
Thanks Soner, I am working with Arvind to get this primitive. I will update once done.

soneryaldiz, 2019-12-20:
Thanks Kunal. I also have concerns about the CMFB_*MOS primitives (seems too generic). Let's discuss the rationale for these primitives in our next team meeting.

kkunal1408, 2019-12-20:
Sure. We can discuss that.

parijatm, 2019-12-26:
Is this a duplicate of #266 or an unrelated issue?

soneryaldiz, 2019-12-27:
Indeed. This is for the NMOS. Also, this primitive shall not have shared diffusion.

kkunal1408, 2020-01-09:
Fixed in #303

kkunal1408, 2020-02-15:
closing this issue, considering it's fixed.

closed 2020-02-15 by kkunal1408