ALIGN-analoglayout/ALIGN-public #285: Sporadic PNR failure on five_transistor_ota

parijatm, 2020-01-03:
**pnr_compiler occasionally fails on five_transistor_ota with the following error message:**

Format Issue 
Please check the net vss in module five_transistor_ota
Especial the pin B in subblock m3

Call to '/ALIGN-public/PlaceRouteHierFlow/pnr_compiler /dataVolume/five_transistor_ota/3_pnr five_transistor_ota.lef five_transistor_ota.v five_transistor_ota.map layers.json five_transistor_ota 1 0' failed with error message:

pnr_compiler: GcellGlobalRouter.cpp:124: GcellGlobalRouter::GcellGlobalRouter(PnRDB::hierNode&, PnRDB::Drc_info&, int, int, const string&): Assertion `0' failed.

**Suggest running multiple trials on five_transistor_ota to reproduce**

parijatm, 2020-01-03:
Log file for last failing run @ https://circleci.com/gh/ALIGN-analoglayout/ALIGN-public/17433

Lastdayends, 2020-01-03:
@parijatm,

Hi Parijatm,

That's the issue as I mentioned in our last meeting. If you look into the run, you can find the flow fails to generate lef file of this design, or pins info are missed in the generated the lef file.

Yaguang

parijatm, 2020-01-03:
Thank you @Lastdayends.

@kkunal1408 are you looking into this already? We didn't go into this during the last meeting since the error went away during a subsequent commit. However it now looks like it is not a systematic bug but only shows up on certain runs.

kkunal1408, 2020-01-05:
@Lastdayends 
Hi Yaguang,
I compared two runs where these are failing. The sporadic failure is due to different automatic constraints generated in two runs. 
Passing constraints: SymmBlock ( {m5_m4} , {m1_m2} , {m0} , {m3} )
Failing constraints: SymmBlock ( {m3} , {m5_m4} , {m1_m2} , {m0} )

Does the order matter for SymmBlock constraints? 
I have uploaded both cases on private repo:
 https://github.com/ALIGN-analoglayout/ALIGN/tree/run_time_v2/PlaceRouteHierFlow/Testcases/DEBUG

Can you please check?
Kunal

Lastdayends, 2020-01-05:
Hi Kunal,

Sure. I can check that.

Yaguang

parijatm, 2020-01-07:
Fixed by #286

closed 2020-01-07 by parijatm