ALIGN-analoglayout/ALIGN-public #342: Current Regressions Issues: Opens, Lots of white space

stevenmburns, 2020-01-23:
2019/1/23

@Lastdayends @parijatm @kkunal1408 

Looking at the current nightly regression I see a couple of issues that could be fixed:

- [ ] Some circuits are generating open when there is plenty of room to route the disconnected  components. Good examples are: FinFET14nm_Mock_PDK/high_speed_comparator and FinFET14nm_Mock_PDK/telescopic_ota_with_bias/3_pnr

- [ ] At least on circuit has a huge amout of white space surrounding the primitives (seems like the terminal are specified to be 20x the primitive size or more away from the primitive. An example here is: FinFET14nm_Mock_PDK/telescopic_ota_with_bias/3_pnr

stevenmburns, 2020-01-23:
Another example in the second catagory is: FinFET14nm_Mock_PDK/switched_capacitor_filter

stevenmburns, 2020-01-23:
Maybe the second category of errors is because capacitors are missing from the layouts.

stevenmburns, 2020-01-23:
The second case is now fix. The first one remains.

stevenmburns, 2020-02-01:
@Lastdayends Are you able to replicate the open (disconnected clk net) in the high_speed_comparator?

Lastdayends, 2020-02-01:
Hi Steve,

Yes. I am looking into it.

Yaguang

stevenmburns, 2020-02-02:
Being addressed by PR #355

closed 2020-02-02 by stevenmburns