ALIGN-analoglayout/ALIGN-public #344: Viewer Issue? + Missing Cap?

parijatm, 2020-01-23:
> @parijatm
> 
> There is one error when I run the master. The Cap_48f_AspectRatio_3x2.json should be under 3_pnr/Results/.
> 
> ERROR align.pnr.checkers:checkers.py:113 /home/yaguang/Desktop/opt/master/work/FinFET14nm_Mock_PDK/single_to_differential_converter/3_pnr/Cap_48f_AspectRatio_3x2.json not found in input_dir
> 
> Also, the caps are still not visiable in the viewer. (In gds, visiable).
> 
> Yaguang

@kkunal1408, @arvuce22 can you take a look at single_to_differential_converter?

@Lastdayends which circuit are you seeing missing caps in the viewer?

This issue may be related to #342 opened by @stevenmburns

parijatm, 2020-01-23:
@kkunal1408 , @arvuce22, @Lastdayends you can hold off on working on this. @stevenmburns and I are doing an initial triage.

parijatm, 2020-01-23:
Addressed by #345. Explanation copied below:

> Caps are a somewhat weird beast as of now since there is custom code to generate capacitor array primitives as part of PlaceRouteHierFlow. This results in the entire .json file being created and put in the Results/ directory instead of the primitives directory where we would usually pick primitives from.
> 
> I have fixed this for now by copying said json files over to the working directory used by PNR to store .json files for intermediate levels of hierarchy. In the long run, we may want to reduce code redundancy and support capacitor array generation as part of primitive generation.

closed 2020-01-23 by parijatm

Lastdayends, 2020-01-23:
Thanks for the update.

Yaguang

Lastdayends, 2020-01-24:
It's linear_equalizer.

Yaguang