ALIGN-analoglayout/ALIGN-public #366: Functionality to generate cells with smaller number of fins.

kkunal1408, 2020-02-18:
Right now cell generator flow doesn't support the creation of cells with 75 fins, the current flow uses the nearest multiple of 12. We need cell generator to draw 6 cells with 12 fins and one cell with 3 fins.
It's important to be LVS clean.

soneryaldiz, 2020-04-24:
@arvuce22 , please see the powertrain_cell example on the linked PR. This example is following the convention we agreed upon (nfin, nf, m):

- nfin : Number of fins in an atomic transistor (equivalent to W in bulk technologies)

- nf : Number of fingers that share diffusion

- m : Number of instances of the transistor specified with nfin and nf

The generated layout for this cell with "-U_mos 4" is shown below and doesn't follow this convention.

[image]

arvuce22, 2020-04-24:
@Yaldiz, Soner <soner.yaldiz@intel.com> I will check this

closed 2020-05-04 by soneryaldiz