ALIGN-analoglayout/ALIGN-public #367: PnR crash due to symmetry constraints.

kkunal1408, 2020-02-19:
Example inverter_v1.
Crash if INV_LVT has a constraint "SymmBlock ( {xm0} , {xm1} )".
The test case can be reproduced by removing "and name not in lib_names" from line 188 in align/compiler.py

kkunal1408, 2020-02-19:
correcting the path of file "align/compiler/compiler.py"

Lastdayends, 2020-02-19:
@kkunal1408 

The SymmBlock ( {xm0} , {xm1} ) breaks one assumption, that the symmetry block should have seem type of master name. In this case, one is Nmos, one is Pmos.

kkunal1408, 2020-02-19:
From my understanding, here I am defining self symmetry of xm0 and xm1, there is no relation between xm0 and xm1.

kkunal1408, 2020-02-19:
And symmetry between xm0 and xm1 will be defined as "SymmBlock ( {xm0, xm1} )"

Lastdayends, 2020-02-24:
@kkunal1408 

There is no such code "and name not in lib_names" in align/compiler/compiler.py

I run current master code without changing compiler.py. It works.

How to reproduce the error case?

kkunal1408, 2020-02-24:
Its on line Line 181: 
 if name not in design_setup['DIGITAL'] and name not in lib_names:

closed 2021-10-30 by kkunal1408