ALIGN-analoglayout/ALIGN-public #370: PnR assertion for string should be removed.

kkunal1408, 2020-02-21:
Virtuoso allows subcircuit names to start from digits, so the generated verilog has module names starting from digits which fails during PnR with the error.
for line 12b_ADC XI0 ( .AVDD(AVDD),****, .AVSS(AVSS)  );

pnr_compiler: Lexer.h:234: void Lexer::error(const string&): Assertion `!failed' failed.

Corresponding design "Santized_12b_ADC_TOP"

stevenmburns, 2020-02-21:
I can do this, but this is not legal verilog syntax. We can make it legal by escaping names properly (add a leading backslash and a trailing space), but this is not currently implemented, and there are many other extensions already implemented that should be escaped as well.
I think the correct thing is to not have this verilog-like file at all, and instead transfer the information with a tailor-made JSON file---I did lose that battle when we had it a year ago.

kkunal1408, 2020-02-21:
The other temporary solution can be that at annotation I prepend it with "align_" (or some other character). 
Moving to a tailor-made JSON file would be fine for me, I can start working on that if it makes more sense. Do we have any example?

stevenmburns, 2020-02-21:
Being addressed by #371

closed 2020-03-18 by kkunal1408