ALIGN-analoglayout/ALIGN-public #382: Power net name corruption

soneryaldiz, 2020-03-18:
If a power net has underscore character in its name, the net name gets corrupted thru the flow. The issue can be reproduced by running the high_speed_comparator example in bug/power_net_name branch:

[image]

Generated Verilog:
[image]

kkunal1408, 2020-03-18:
It's been fixed in, "Fix_timeout" branch. Please try and let me know, then I will merge this branch before pushing more updates.

soneryaldiz, 2020-03-19:
Hi Kunal, thank you for the prompt fix. The net names are preserved now as the snapshot below shows. I pushed a commit to revise this test case accordingly.

[image]

closed 2020-04-17 by kkunal1408