ALIGN-analoglayout/ALIGN-public #408: Overriding automatic constraints

meghna09, 2020-04-08:
I was wondering if there's a switch to override automatic constraints and use only manual constraint files. I'm not sure where this switch is available.

The branch vco_testcase has an example (vco_dtype_12_hierarchical_res_constrained).  It has both manual (user-specified) + automatic constraints added to end it by the time it passes through the flow.

kkunal1408, 2020-04-08:
Right now if there is any SymmNet constraint, flow won't generate SymmNet, same with SymmBlock. 
I will improvise to make it based file-based checks.

meghna09, 2020-06-18:
Currently, annotation and constraint generation are linked. If a block is defined as digital/unconstrained in the setup file, no annotation is carried out either. There may be a need to separate the two, keep the tool defined annotation and user-defined constraints in some cases as an option. I think the file-based check may be a solution to this.

kkunal1408, 2020-06-18:
1. Digital--> No annotation is done + user-defined constraints are allowed
2. Symmblock constraint--> Annotation + only user-defined constraints will be applied
3. In the last discussion, we agreed to add constraints even in the case of user-defined constraints. 
Moving to file-based check will affect point three, if you want we can add a separate switch in the config file for annotation and constraint generation. 
Will that be fine?

closed 2021-04-26 by meghna09