ALIGN-analoglayout/ALIGN-public #420: Ability to provide topology specifications

soneryaldiz, 2020-04-16:
I want to provide specifications to topology identification in the form of:
- exclude list of devices from identification
- enforce mapping list of devices to a primitive

This allows user to bypass false identifications (e.g., do not identify diff pair) or to enforce design intention (e.g., identify pseudo diff pair as inverter or identify decap pair as replica diff pair).

kkunal1408, 2020-04-17:
1. To exclude any device from identification you can specify any of its net as CLK in the setup file.
2. To add any new primitive to the library you can add them as spice format in user_defined.sp file
Please let me know if you have any other suggestions to improve these workarounds.

soneryaldiz, 2020-04-29:
Thanks Kunal. I will try your suggestions. Designating a net a CLK might be too restrictive as it would impact all transistors. My second bullet is slightly different: Suppose there are two transistors m1 and m2 in the netlist and they are either not identified or identified as decaps. I want to enforce that these two transistors are recognized as a differential pair. One example where this happens is noise cancellation circuits. This capability also allows user to guide/ease topology identification for more complex/novel circuits.

soneryaldiz, 2021-04-27:
Closing this issue as #647 will provide this.

closed 2021-04-27 by soneryaldiz