ALIGN-analoglayout/ALIGN-public #423: powertrain_binary and powertrain_thermo failures

soneryaldiz, 2020-04-22:
ALIGN throws assertion for powertrain_binary and powertrain_thermo examples. Reproducible on master branch.

[image]

[image]

Lastdayends, 2020-04-23:
@soneryaldiz 

The terminal of the module is illegal. The terminal name here is:

module powertrain_binary ( on_d[0], on_d[1], on_d[2], on_d[3], on_d[4], on_d[5], vout ); 

on_d[0], '[' is illegal here.

Yaguang

Lastdayends, 2020-04-23:
@stevenmburns 

Hi Steve,

In Soner's case, an illegal terminal name appears in the verilog file. Should we make that illegal name illegal or just change the terminal name to other illegal name?

module powertrain_binary ( on_d[0], on_d[1], on_d[2], on_d[3], on_d[4], on_d[5], vout );

on_d[0], '[' is illegal here.

Yaguang

Lastdayends, 2020-04-23:
@stevenmburns 

Sorry for the typo. I mean, should we make that illegal name legal or just change the terminal name to other legal name?

Yaguang

stevenmburns, 2020-04-23:
This is easy to fix in the Verilog parser. I’ll add it to the PR.

Lastdayends, 2020-04-23:
@stevenmburns 

Thanks!

Yaguang

closed 2020-04-29 by soneryaldiz