ALIGN-analoglayout/ALIGN-public #424: unity_gain_buffers failure

soneryaldiz, 2020-04-22:
unity_gain_buffers circuit fails on master branch:

[image]

soneryaldiz, 2020-04-22:
@kkunal1408 , @arvuce22 , @parijatm : This test case uses non-stacked, stack of 2 and stack of 4 transistors. It also uses shorter transistors with fewer fins. I will elaborate more during our virtual F2F.

soneryaldiz, 2020-05-13:
Hi @kkunal1408 , the topology identification looks good on bug/compiler branch:
[image]

However, there are still two issues in the generated netlist (i.e., *_blocks.sp):
1) This netlist does not reflect the stacked transistors.
2) There are dangling body terminals in the unity_gain_buffers subcircuit which should have been connected to a power net.

kkunal1408, 2020-09-22:
Will test this with issue #449 ?

soneryaldiz, 2020-09-24:
Hi @kkunal1408 , I updated the unity_gain_buffers netlist in this PR #483 per the new transistor parameter convention. Below is the generated v file:
```
module unity_gain_buffers ( vbias_an, vo_hs, vo_ls, vref_hs, vref_ls ); 
input vbias_an, vo_hs, vo_ls, vref_hs, vref_ls;

Switch_NMOS_w4_nf1.0_m1.0_n12_X2_Y1_ST2 xmn43 ( .B(vss), .D(vo_ls), .G(v_p1), .S(vss) ); 
Switch_NMOS_w4_nf1.0_m1.0_n12_X2_Y1_ST2 xmn41 ( .B(vss), .D(vbias4), .G(vbias_an), .S(vss) ); 
Dcap_NMOS_w4_nf1.0_m1.0_n12_X2_Y1_ST2 xmn3 ( .B(vss), .G(vbias_an), .S(vss) ); 
Switch_NMOS_w4_nf1.0_m1.0_n12_X3_Y1_ST2 xmn2 ( .B(vss), .D(vcom_n), .G(vbias_an), .S(vss) ); 
DCL_NMOS_w4_nf2.0_m2.0_n12_X11_Y1 mmn42 ( .B(vss), .D(vbias_m), .S(vbias4) ); 
DCL_PMOS_w4_nf2.0_m2.0_n12_X11_Y1 mmp33 ( .B(vcc), .D(vbias_m), .S(vbias_n1) ); 
Switch_PMOS_w4_nf1.0_m1.0_n12_X2_Y1_ST2 xmp7 ( .B(vcc), .D(vo_hs), .G(v_n2), .S(vcc) ); 
SCM_NMOS_w4_nf1.0_m1.0_n12_X3_Y1_ST2 xmn0_xmn1 ( .B(vss), .DA(v_p2), .DB(v_p1), .S(vss) ); 
SCM_PMOS_w4_nf1.0_m1.0_n12_X3_Y1_ST2 xmp28_xmp22 ( .B(vcc), .DA(v_n1), .DB(v_n2), .S(vcc) ); 
SCM_PMOS_w4_nf1.0_m1.0_n12_X3_Y1_ST2 xmp34_xmp10 ( .B(vcc), .DA(vbias_n1), .DB(vcom_p), .S(vcc) ); 
DP_NMOS_w4_nf1.0_m1.0_n12_X7_Y1_ST2 xmn23_xmn22 ( .B(vss), .DA(v_n2), .DB(v_n1), .GA(vref_hs), .GB(vo_hs), .S(vcom_n) ); 
DP_PMOS_w4_nf1.0_m1.0_n12_X7_Y1_ST2 xmp8_xmp9 ( .B(vcc), .DA(v_p2), .DB(v_p1), .GA(vo_ls), .GB(vref_ls), .S(vcom_p) ); 

endmodule
```

Several devices in this netlist are stack=4 (e.g., xmn41, xmn3). However, they are all printed as *_ST2 above.

kkunal1408, 2020-12-02:
@soneryaldiz Is this still an issue?

closed 2021-10-30 by kkunal1408