ALIGN-analoglayout/ALIGN-public #439: current mirror definition

meghna09, 2020-06-18:
SCM represents two transistors in a current mirror with a 1:1 ratio. Sometimes, the two transistors have a different sizing maybe fingers are in the ratio 1:4, and in this case a different primitive needs to be called. Currently, the SCM primitive cell is called even when the transistor sizes are different.

Testcase will be added if needed.

kkunal1408, 2020-06-18:
@arvuce22: do we have separate cells for two scenarios?
@meghna09: How are both scenarios different if we consider common centroid placement?
Can you provide a test case for this?

meghna09, 2021-04-26:
@kkunal1408 there are two separate cells. For a current mirror with 1:1 ratio - SCM needs to be used. For a current mirror with 1:n ratio CM needs to be used.

@arvuce22 CM parameter definition is not clear and needs to be modified.

arvuce22, 2021-11-17:
We are trying to resolve this issue using PR #870

closed 2021-11-18 by kkunal1408