ALIGN-analoglayout/ALIGN-public #442: Still Shorts in some FinFET Regression Circuits

stevenmburns, 2020-06-25:
@kkunal1408 @Lastdayends I'm seeing shorts in "adder".
You can duplicate the results using:
```
pytest -vv --runnightly  -k 'adder and FinFET' -- tests/integration/test_all.py
```
Here is the visualization:
First without the vin nets:
[image]
Then with the vin nets:
[image]

It could also be a viewer error. Should the signals C0/dummy_gnd_minus and vin be connected together? I don't see the dummy unit cells connected in the netlist. Should they be connected to gnd? Where should that be done?

```text
module adder ( n1, n2, vin, vout ); 
input n1, n2, vin, vout;

Switch_NMOS_n12_X2_Y2 xI0|MN0 ( .B(b), .D(vout), .G(vbn1), .S(vgnd) ); 
Switch_PMOS_n12_X2_Y2 xI1|MP0 ( .B(b), .D(vout), .G(vbp1), .S(vps) ); 
Res_20000 R0 ( .PLUS(n1), .MINUS(vbn1) ); 
Cap_48f C0 ( .PLUS(vbn1), .MINUS(vin) ); 
Res_20000 R1 ( .PLUS(n2), .MINUS(vbp1) ); 
Cap_48f C1 ( .PLUS(vbp1), .MINUS(vin) ); 
Res_500 R2 ( .PLUS(vout), .MINUS(vps) ); 

endmodule

`celldefine
module global_power;
supply0 vgnd;
supply1 vps;
endmodule
`endcelldefine
```

Lastdayends, 2020-06-25:
@stevenmburns @kkunal1408 

The the signals C0/dummy_gnd_minus is the pin for dummy cap in C0. They should be connected to global_power vgnd, not vin.

The dummy_gnd_minus is not shown in the netlist file, and it's generated by the common centroid capacitor code.  However, there is a mismatch between the pins of C0 in verilog and in our database.  In verilog, there are only vbn1 and vin, while in our database, there are vbn1, vin, C0/dummy_gnd_minus and C0/dummy_gnd_plus.

C0/dummy_gnd_minus is be a power pin, not a signal pin. The power routing should be done in the power routing.

There are two options to avoid the issue:
1. ignore the C0/dummy_gnd_minus, because it does not exist in verilog.
2. add some assumptions to connect the C0/dummy_gnd_minus. For example, C0/dummy_gnd_minus is always connected to the global_power vgnd.

stevenmburns, 2020-06-25:
We seem to need to do two things. 1) connect these dummy_gnd signals to actual gnd, and 2) keep these signals from connecting to (shorting with) anything else. I can see the 1) can be fixed by setting up the problem correctly (telling the power router to make the connection.) It seem that 2) might be harder. Does the router not see these wires as obstructions? Do we need to create a proper LEF file so the router knows that it can't connect to these spots?

Lastdayends, 2020-06-25:
@stevenmburns 
1) We need some internal assumption to connect these dummy_gnd signals to actual gnd. Since the name of actual gnd is controlled by verilog and the name of dummy_gnd_signal is hard-coded (can not know from verilog), we can assume  dummy_gnd_signal is always connected to the supply0 in verilog file.
2) I can add those dummy_gnd_signal pins as obstructions internally, so LEF file is not necessary.

stevenmburns, 2020-06-25:
I think supply0 is a good start. @kkunal1408 Should we add these dummy_gnd signals to the verilog?

kkunal1408, 2020-06-25:
The number of dummies are decided by cap generator, (dependent on aspect
ratio) so during verilog generation it will be a floating port. I think
handling during PnR should be fine, connecting it to supply0. Only concern
I find, if these passives get connected to analog ground. But anyways we
are not handling multiple power/gnd rails right now.

Once we start handling multiple ground, we can add this information in the
constraint file.
Current constraint format: CC ( {c0_c4} , {5,5} , {Cap_12f} , {nodummy} )
// default is dummy of same type as Cap_12f
We can modify this to something like: CC ( {c0_c4} , {5,5} , {Cap_12f} ,
{dummy=Cap_2f, gnd=vss} )

Thanks
Kunal

Lastdayends, 2020-06-28:
@stevenmburns @kkunal1408 

Now the shorts issue is solved in the branch, 
https://github.com/ALIGN-analoglayout/ALIGN-public/tree/fix_bug/cap_pin

4 opens are reported by Intel checker, however, in our RouterReport, every routing path has been found.

Lastdayends, 2020-06-28:
@meghna09 

As far as I remembered, the same cases happen to you that opens are reported by Intel checker while they do not exist.

stevenmburns, 2020-07-09:
Shorts were fixed in #445 and #448 for the "adder" testcase. There are issues with "switched_capacitor_filter". I'll close this and open a new issue for that block.

closed 2020-07-09 by stevenmburns