ALIGN-analoglayout/ALIGN-public #443: DRC Opens still exist in some regression examples

stevenmburns, 2020-06-25:
@Lastdayends @kkunal1408 I see a DRC open in the same testcase as I found shorts (#442 adder).
On the signal vin, there are m4 wires going vertically (why?) and connecting to a horizontal m4 wire (upper left corner). The vertical wire is off-grid (actually wrong way). Is this something that we want to do? I see the same thing for m3 (at the bottom right edge.)

[image]

stevenmburns, 2020-06-26:
@Lastdayends @kkunal1408 Can you confirm that this is a error? Where is the veritical metal4 coming from?

Lastdayends, 2020-06-26:
@stevenmburns 

I will confirm.

Lastdayends, 2020-06-26:
@stevenmburns 

That was created by our detailed router. I have turned off the code to create off-grid routing metals.

kkunal1408, 2020-06-27:
Will it create opens for them now or what will it do?

Lastdayends, 2020-06-27:
@stevenmburns @kkunal1408 

Bidirectional routing is removed.

stevenmburns, 2020-07-09:
Opens were fixed in #445 and #448 for the "adder" testcase. There are issues with "switched_capacitor_filter". I'll close this and open a new issue for that block.

closed 2020-07-09 by stevenmburns