ALIGN-analoglayout/ALIGN-public #446: Add a user defined cell flow

Jiteshp01, 2020-06-28:
We need a flow where user defined cell (like a black box layout of inductor or custom capacitor) can be added easily.
Currently, we use back-end changes to suit the flow.

kkunal1408, 2020-10-06:
Checks which need to be done for any user-defined cell for PnR to run properly:
- Height of the external cell should be multiple of M2 pitch
- Pin width should be>= 2-time metal pitch defined in layers.json
- Width of pin >= 2 times M1/M3 pitch
- Height of pin >= 2 times of M2 pitch
- Name of cells in Verilog, map, LEF, gds.json should match

parijatm, 2022-01-21:
Off-grid issues makes this a PDK specific task

@srini229 already has a script to generate pin /  blockage information from GDS. Follow up items:
- [ ] Check this into master under scripts
- [ ] Please also add the generated output and include that in the flow
- [ ] Apply to multiple PDKs and identify common code

Note: We already have some hand written ones here:
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/align/pdk/finfet/digital.py
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/align/pdk/finfet/resistor.py

soneryaldiz, 2022-11-03:
@srini229 will look into this.

soneryaldiz, 2023-02-09:
Resolved by the Hanan grid router for now. See align/gdsconv/gds2primitive.py

closed 2023-02-09 by soneryaldiz