ALIGN-analoglayout/ALIGN-public #447: handling multivt cells in compiler

kkunal1408, 2020-06-30:
handling multiple models of transistors [lvtnfet,lvtpfet,slvtnfet,nmos_rvt,pmos_rvt,nmos,pmos]

kkunal1408, 2020-06-30:
Cell generator commands
e.g., python Align_primitives.py -p DP_NMOS -b test -n 12 -X 2 -Y 2 -v HVT
Other options are RVT, HVT, or LVT (RVT is the default).
Modifying all netlists to these 4 formats:
add model file for different transistors
[lvtnfet,lvtpfet,slvtnfet,hvtnfet,hvtpfet,nmos,pmos]

soneryaldiz, 2020-07-21:
@kkunal1408 , how about we pass the model name from the netlist to the primitive (mos) generator as the vt_type? Here are few examples:
`m1 d g s b n nfin=3`       => vt_type = n
`m2 d g s b nhvt nfin=3`  => vt_type = nhvt
`m2 d g s b plvt nfin=3`   => vt_type = plvt

kkunal1408, 2020-09-22:
same as issue #479

closed 2020-09-22 by kkunal1408