ALIGN-analoglayout/ALIGN-public #450: Shorts (and other errors) in switched_capacitor_filter

stevenmburns, 2020-07-10:
@Lastdayends @kkunal1408 With the latest fixes, there are still routing errors (shorts) in switched_capacitor_filter. Here is the first error:

This is just net66.
[image]

This is net66 and vinn:
[image]

There are many shorts (1208). Here are the first 10:
```text
SHORT CONNECTPAIR M1 ([28144, 13992, 28176, 16500], 'c5_c2/MINUS2') ([28144, 16428, 28176, 16500], 'net64')
SHORT CONNECTPAIR M1 ([28144, 16932, 28176, 19440], 'net66') ([28144, 19368, 28176, 19440], 'vinn')
SHORT CONNECTPAIR M1 ([28064, 972, 28096, 3480], 'c0_c4/MINUS2') ([28064, 3408, 28096, 3480], 'net72')
SHORT CONNECTPAIR M1 ([28064, 3912, 28096, 6420], 'c0_c4/MINUS2') ([28064, 6348, 28096, 6420], 'net72')
SHORT CONNECTPAIR M1 ([28064, 6852, 28096, 9360], 'net60') ([28064, 9288, 28096, 9360], 'net63')
SHORT CONNECTPAIR M1 ([28064, 9792, 28096, 12300], 'c0_c4/MINUS2') ([28064, 12228, 28096, 12300], 'net72')
SHORT CONNECTPAIR M1 ([25024, 3912, 25056, 6420], 'net60') ([25024, 6348, 25056, 6420], 'net63')
SHORT CONNECTPAIR M1 ([25024, 6852, 25056, 9360], 'net60') ([25024, 9288, 25056, 9360], 'net63')
SHORT CONNECTPAIR M1 ([25024, 9792, 25056, 12300], 'c0_c4/MINUS2') ([25024, 12228, 25056, 12300], 'net72')
SHORT CONNECTPAIR M1 ([25104, 13992, 25136, 16500], 'net66') ([25104, 16428, 25136, 16500], 'vinn')
```

Lastdayends, 2020-07-10:
@stevenmburns 

Ok. I will check it.

Lastdayends, 2020-07-10:
@kkunal1408 @stevenmburns 

There is a bug in the verilog file.

The common centroid capacitor code needs to know the pair of pins.

Currently it is:

Cap_cc c8_c3 ( .MINUS0(voutn), .MINUS1(voutn), .PLUS0(vss), .PLUS1(net66) ); 

In this case, is .MINUS0(voutn) and .PLUS0(vss) a pair? Or .MINUS0(voutn) and .PLUS1(net66) a pair?

I would suggest: 

Cap_cc c8_c3 ( .MINUS0(voutn), .PLUS0(vss), .MINUS1(voutn), .PLUS1(net66) ); 

So that it is clear that .MINUS0(voutn) and .PLUS0(vss) is a pair. And .MINUS1(voutn), .PLUS1(net66) a pair.

stevenmburns, 2020-07-10:
I think it is clear that MINUS0 and PLUS0 make up one pair, and MINUS1 and PLUS1 make up the other pair. We can use the formal pin names instead of relying on the ordering of the parameters.

Lastdayends, 2020-07-10:
@stevenmburns 

Should we distinguish the pair based on pin NAME+number? If the name are paired and number is same, then make pins a pair?

Then 'MINUS' and 'PLUS' are probably hard-coded as key of name in the code.

stevenmburns, 2020-07-10:
Yes, that makes sense to me. Should this be fixed in the Verilog parser or in the cap_placer?

Lastdayends, 2020-07-10:
@stevenmburns 

In the cap_placer, I can fix that.

stevenmburns, 2020-07-10:
@Lastdayends @kkunal1408 @arvuce22 We can probably enumerate the possible formal names you will get from the frontend.
Seems like we have "PLUS", "MINUS", "dummy_gnd_PLUS", "dummy_gnd_MINUS", "PLUS1", "MINUS1", "PLUS2", "MINUS2". Any others?

Lastdayends, 2020-07-10:
"PLUS0" and "MINUS0".

Lastdayends, 2020-07-10:
@stevenmburns 

[image]

Seems the viewer do not get the right cap here. 

The shown cap should be connected to vin. And the caps connected to net66 is missed.

Lastdayends, 2020-07-10:
I mean:

[image]

stevenmburns, 2020-07-11:
Is this in a PR so I can try it out?

Lastdayends, 2020-07-11:
@stevenmburns 

Yes. Those caps are in PR.

stevenmburns, 2020-07-11:
Is the PR #453 ?
If it is there are still  1208 shorts. The first short is between vss and net66.

Lastdayends, 2020-07-11:
@stevenmburns 

Yes. It is #453.

Lastdayends, 2020-07-11:
@stevenmburns 

Ok. I will check.

stevenmburns, 2020-07-12:
The .json file generated by the cap_placer has six terminals: "MINUS0", "PLUS0", "MINUS1", "PLUS1", and "MINUS2", "PLUS2", but the verilog file has only four: "MINUS0", "PLUS0", and "MINUS1", "PLUS1". That could be the problem. 
This is the Python version of the PnRDB lookup code which is probably only finding the four values:
```python
    fa_map = {}
    for n in itertools.chain( hN.Nets, hN.PowerNets):
        for c in n.connected:
            if c.type == 'Block':
                cblk = hN.Blocks[c.iter2]
                blk = cblk.instance[cblk.selectedInstance]
                block_name = blk.name
                master_name = blk.master
                pin = blk.blockPins[c.iter]
                formal_name = f"{blk.name}/{pin.name}"
                assert formal_name not in fa_map
                fa_map[formal_name] = n.name

            else:
                term = hN.Terminals[c.iter]
                terminal_name = term.name
                assert terminal_name == n.name
```

Maybe you updated the code that generates the rectangles in cap_placer (and spits out .gds.json) and didn't update the code that generates "primitive"-style json.

stevenmburns, 2020-07-12:
Here are the netName fileds in a couple for representativein .json files:
```text
(general) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ egrep netName work/FinFET14nm_Mock_PDK/switched_capacitor_filter/3_pnr/Cap_cc_AspectRatio_2x3.json  | sort | uniq -c
     31             "netName": "MINUS0",
    429             "netName": "MINUS1",
    393             "netName": "MINUS2",
     31             "netName": "PLUS0",
    429             "netName": "PLUS1",
    393             "netName": "PLUS2",
(general) smburns@smburns-XPS-13-9370:~/DARPA/ALIGN-public$ egrep netName work/FinFET14nm_Mock_PDK/switched_capacitor_filter/3_pnr/Cap_cc_AspectRatio_2x4.json  | sort | uniq -c
    291             "netName": "dummy_gnd_MINUS",
    291             "netName": "dummy_gnd_PLUS",
     31             "netName": "MINUS0",
    429             "netName": "MINUS1",
    393             "netName": "MINUS2",
     31             "netName": "PLUS0",
    429             "netName": "PLUS1",
    393             "netName": "PLUS2",
```

Lastdayends, 2020-07-12:
@stevenmburns 

This bug was fixed in #453.

Lastdayends, 2020-07-12:
Now there are 1 open and 5 minlength bug.

Lastdayends, 2020-07-12:
@stevenmburns 

Now switched capacitor filter is drc clean.

closed 2020-07-13 by stevenmburns