ALIGN-analoglayout/ALIGN-public #454: primitive caps of size 1nF are not getting generated

kkunal1408, 2020-07-13:
Primitive generator takes very long time (>30 min) for generating 1nF cap in sc_dc_dc_converter.  Can't generate these caps using common centroid methodology also.

kkunal1408, 2020-07-13:
max cap size is ~10pF

arvuce22, 2022-01-20:
@kkunal1408 This needs a discussion: whether that big single cap is used (generally unit cells are connected to implement a larger cap; PDKs does not support)

kkunal1408, 2022-01-20:
@arvuce22 sure. Then can we have a parameter from PDK layers.json on what size it is supported?

arvuce22, 2022-01-20:
@kkunal1408 yes, we can add max size for caps and resistance (similar to we have for nfins: "max_nfin_mos")

parijatm, 2022-01-21:
the max_* parameters can be used as a check (throw an assertion if something higher arrives) but this should be a task handled by the person writing the netlist.

If any automatic breaking up is needed, should be done as a netlist pre-processing step.

closed 2022-01-22 by kkunal1408