ALIGN-analoglayout/ALIGN-public #474: Use pytest 6.0.0

parijatm, 2020-09-02:
pytest version is currently fixed to 5.4.3 (#473). May want to upgrade to pytest 6.0.0 and fully root-cause INTERNALERROR issue at some point.

parijatm, 2020-09-02:
I have root-caused the INTERNALERROR to a pytest bug that is exercised when you generate junit.xml and use --tb=line.

Can we drop the --tb=line? Tagging both @kkunal1408 @stevenmburns since I am not sure who added it.

The other option would be to leave it as it is now and file a bug against pytest.

parijatm, 2020-09-03:
Also filed https://github.com/pytest-dev/pytest/issues/7707

kkunal1408, 2020-09-03:
I added the tb=line so that only the DRCs count gets populated when the assert statement for DRC count=0 fails. Without this command, it dumps a long log (terminating after a fixed length) instead of a concise view thus affecting the readability of the report.

parijatm, 2020-09-03:
Got it. Let's stick with pytest 5.4.3 until we get a 6.0 fix in that case. 

The bug has been acknowledged and is being addressed in an active pull request.

parijatm, 2020-09-19:
Bug has been addressed in pytest 6.0.2. ALIGN will be contoured with latest pytest once PR #449 is merged with master.

closed 2020-09-19 by parijatm