ALIGN-analoglayout/ALIGN-public #485: Transistor parameters in complex primitives

soneryaldiz, 2020-09-27:
The test circuit is test/compiler/test_circuits/intel_circuit4.sp:

1. CMB_PMOS_2: Transistor sizing within the subcircuit does not match netlist. M0_M1 (m=32) mapped to xmp11 (m=16) and xmp12 (m=32).

2. CMB_PMOS_2: SCM_PMOS_nfin4_m32_n12_X11_Y1_ST4 M0_M1 is missing vt_type (real_inst_type is blank in the log).

3. INV_LVT: xmn1 is using a different vt_type yet both xmn1_xmp1 and xmn0_xmp0 are mapped to the same primitive. What this implies is we need to make sure complex primitives need to be unique (aware of transistor parameters).

kkunal1408, 2020-10-06:
fixed in branch transistor model need verification.

soneryaldiz, 2020-10-09:
Verified and PR is merged.

closed 2020-10-09 by soneryaldiz