ALIGN-analoglayout/ALIGN-public #494: Wrong sub circuit name in user template

kevingo, 2020-11-23:
The above sub circuit in [user template](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/align/config/user_template.sp) should be ended with `LSB_NMOS_7` instead of `LSB_NMOS_2` ?

```
.subckt LSB_NMOS_7 DA DB DC DD DE DF DG DH SA SB SC SD SE SF SG SH
**UT austin  array BUFFER_VREFP2
M0 (DA DA SA B) NMOS w=w l=90n
M1 (DB DA SB B) NMOS w=w l=90n
M2 (DC DA SC B) NMOS w=w l=90n
M3 (DD DA SD B) NMOS w=w l=90n
M4 (DE DA SE B) NMOS w=w l=90n
M5 (DF DA SF B) NMOS w=w l=90n
M6 (DG DA SG B) NMOS w=w l=90n
M7 (DH DA SH B) NMOS w=w l=90n
.ends LSB_NMOS_2
```

kkunal1408, 2020-12-02:
Thanks, @kevingo for pointing it out, the "end" statement should have LSB_NMOS_7 (though it should not affect the annotation as the names are identified based on ".subckt" line. 
I will update this in the next merge.

kkunal1408, 2021-11-08:
Fixed in latest master.

closed 2021-11-08 by kkunal1408