ALIGN-analoglayout/ALIGN-public #532: Use C++ coverage metrics to remove dead code for PnR

stevenmburns, 2021-02-02:
@Lastdayends @854768750 We have lcov running in CI and it shows that a few files are not being used much or at all anymore.
Here is a screen shot from a recent run:

[image]

It shows that "PnRDB/HardDesignRule.cpp" and "PnRDB/ReadDesignRule.cpp" are likely not used anymore (at least their usage is not demonstrate during CI.)

We should remove this files that are no longer needed.

There are others as well. In the Placer there is "Aplace.cpp", "Aplace.h", "ConstGraph.cpp", probably "ConstGraph.h", a lot of "Placer.cpp".

To look into this in more detail, click on Details (circled in red below):
[image]

then ARTIFACTS (circled in red below):
[image]

Then coverage-reports/PlaceRouteHierFlow/htmlcov/index.html (underlined in red below):
[image]

Then pick your directory of interest, e.g., PlaceRouteHierFlow/PnRDB
[image]

854768750, 2021-02-02:
@stevenmburns  Some files may be used later although they are not used now. Does comment the unused lines with `/** **/` work (increase line coverage)?

854768750, 2021-02-02:
@stevenmburns  Is this issue related to failed nightly? Is it going on in https://github.com/ALIGN-analoglayout/ALIGN-public/tree/fix/nightly?

parijatm, 2022-01-21:
Let's try to exceed 60% code coverage by 1/28/2022.

Try to split up into multiple PRs.

closed 2022-02-01 by 854768750