ALIGN-analoglayout/ALIGN-public #540: Port labels are missing in python GDS

meghna09, 2021-02-10:
The port labels/pins are present in the C++ GDS but are missing in the python GDS. This is required to get an LVS clean layout and to carry out extraction and simulation. (This wasn't present the last time I checked, not sure if it has been updated)

stevenmburns, 2021-02-10:
@meghna09 I'll look into it. Do you have a testcase we should be using as an example.

meghna09, 2021-02-10:
@stevenmburns @Lastdayends 
For this test case (telescopic OTA)
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/examples/telescopic_ota/telescopic_ota.sp

The top-level ports "d1 vdd vinn vinp vss vbiasn vbiasp1 vbiasp2 voutn voutp" need to have port labels (text) be displayed on the label layer of the corresponding metal layer.

For example, if the net d1 is on M2, it should have a text label "d1" displayed on the layer with GdsLayerNo 15 and GdsDatatype 4.

"Layer": "M2",
"GdsLayerNo": 15,
"GdsDatatype": {"Draw":0, "c1":1, "c2":2, "Pin":3, "Label":4, "Blockage":5}

stevenmburns, 2021-02-11:
Do you care where these labels are attached to the layout?
I assume you only need it on the top most layer. Can I put them in the middle (or corner) of an arbitrary rectangle on this layer?

meghna09, 2021-02-11:
The router currently prints these pins and their location information in the C++ GDS. The same information can be reused. @Lastdayends can indicate where that information is.

Otherwise, the topmost layer, at the middle of an arbitrary rectangle sounds good.

stevenmburns, 2021-02-11:
@Lastdayends @meghna09 I found out where this information is added to the C++ .gds.json file. I can just do the same thing.

Lastdayends, 2021-02-11:
@stevenmburns 

The label is current printed at the 1st metal of the terminal. The code can be found from line 500 - 523 of
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/PlaceRouteHierFlow/PnRDB/WriteJSON.cpp

soneryaldiz, 2021-02-11:
> Do you care where these labels are attached to the layout?
> I assume you only need it on the top most layer. Can I put them in the middle (or corner) of an arbitrary rectangle on this layer?

Yes, the label locations might matter for commercial extract flows and impact the parasitics. It is preferable to have them in the middle.

stevenmburns, 2021-02-12:
Starting a fix in PR #543 . Let's move the discussion there.

soneryaldiz, 2022-11-03:
@meghna09 , is this still needed?

meghna09, 2022-11-10:
@soneryaldiz, I think we have a fix for this at our end so this is not needed.

closed 2022-11-10 by meghna09