ALIGN-analoglayout/ALIGN-public #545: Codacy isn't running anymore

stevenmburns, 2021-02-13:
@parijatm @Lastdayends @arvuce22 @kkunal1408 @soneryaldiz 
Does anyone know when this stopped running? Did someone stop it on purpose?

parijatm, 2021-02-14:
Looking at the commits, it looks like Codacy stopped working sometime around November 1st 2020. Don't think it was intentional @stevenmburns (or at least I don't see any relevant commits).

The .codacy.yml file seems to have been last edited in January 2020 so it is unlikely to be the cause of the issue. I suspect @ALIGN-analoglayout will need to redo the github integration.

kkunal1408, 2021-12-17:
@stevenmburns I don't think .codacy.yml is a requirement. I have created a ticket with the Codacy team. Have we tried reintegrating? 

Optional: I tried doing it for one of my personal repositories (https://github.com/kkunal1408/ota_data) and it worked smoothly. This may not be an issue but the link format for the badge seems a little different.

kkunal1408, 2021-12-17:
@stevenmburns I emailed the Codacy team about this. I forwarded the reply I got.

parijatm, 2022-01-21:
Running static checks using flake8 locally.

Codacy does do more linting on the C++ side so it would be nice to have.

closed 2022-11-03 by soneryaldiz