ALIGN-analoglayout/ALIGN-public #555: Matser does not work for tcsh but works for bash.

kkunal1408, 2021-02-21:
On tcsh shell PnR gives a segmentation fault.

kkunal1408, 2021-02-22:
The error is during detail router.

PnR.router.Router.RouteWork INFO : GcellGlobalRouter: telescopic_ota
PnR.router.Router.RouteWork INFO : GcellDetailRouter: telescopic_ota
/bin/bash: line 1: 71483 Segmentation fault      (core dumped)

kkunal1408, 2021-02-22:
Last few lines from LOG file in DEBUG mode
2021-02-22 11:15:49 PnR.router.GcellDetailRouter.ConvertRect2GridPoints DEBUG : Enter converter
2021-02-22 11:15:49 PnR.router.GcellDetailRouter.ConvertRect2GridPoints DEBUG : rect info 0 4768 6816 4832 6960
2021-02-22 11:15:49 PnR.router.GcellDetailRouter.ConvertRect2GridPoints DEBUG : converter check point 2

No more information is shown after this and just shows a segmentation fault.

parijatm, 2021-02-22:
Hi @kkunal1408 just want to make sure I understand the issue. For the same design, same machine, same installation method, you are getting segmentation fault in tcsh and not bash?  Or is ALIGN built using install.sh behaving differently from ALIGN built using install_tcsh.sh?

+ @stevenmburns as he did fixed multiple segmentation faults when integrating pybind11 based PnR.

kkunal1408, 2021-02-22:
Yes ALIGN built using install_tcsh.sh is behaving differently than install.sh. Though there is no error shown during installation on our internal servers.
LOG file using install_tcsh.sh does not have info on why is it happening. Here is the log file (LOG1)
https://drive.google.com/file/d/1DFpMGz0H4q3zLsn97XzcKb_G0ZwIT2aR/view?usp=sharing

This is the log file (LOG2) using install.sh
https://drive.google.com/file/d/10SJSRkir_Wqj2CPFep771caXVpB7XCYr/view?usp=sharing

These are the set of line  in LOG2 which come just after the last lone of LOG1
2021-02-22 11:32:12 PnR.router.Grid.setSrcDest DEBUG : Router-Info: detecting source- 0
2021-02-22 11:32:12 PnR.router.Grid.setSrcDest DEBUG : Router-Info: detecting checkpoint1 0

I think this section can have the issue.

srini229, 2021-02-22:
Hi Kunal,
I was trying to attach debugger by using -g build to see where it was crashing. It started finishing without crashes after that. I removed the -O3 to build the router alone and there were no crashes.

kkunal1408, 2021-02-23:
Based on Ramprasath debug. " The crash seems to happen in the routine "RawRouter::InsertPlistToSet_x". The insertion seems trivial std::set insertion but some memory corruption elsewhere may be causing this. It is constantly crashing at this point always."

kkunal1408, 2021-03-09:
For our server machines  (NAME="Red Hat Enterprise Linux Workstation" VERSION="7.9 (Maipo)" )

The workaround of removing -O3 works for small designs but does not work for large designs such as switched_capacitor_filter.

parijatm, 2021-04-08:
Can we please rename this issue to reflect the actual bug?

srini229, 2021-04-08:
This isn't happening anymore. The runs have been fine after the recent build process change to pip.

parijatm, 2021-04-08:
Fixed by #572.

closed 2021-04-08 by parijatm