ALIGN-analoglayout/ALIGN-public #560: OrderBlocks for subcircuit causes segmentation fault

soneryaldiz, 2021-02-26:
SymmetricBlocks constraint not satisfied.

Branch: fix/const
Example: schematic2layout.py $ALIGN_HOME/examples/ring_oscillator -c
Result:
```
PnR.PnRDB.PnRdatabase.CheckinHierNode INFO : CheckinHierNode
PnR.toplevel INFO : Main-Info: complete node 2
PnR.toplevel INFO : Main-Info: start to work on node 1
PnR.toplevel INFO : Start placement: ring_oscillator_stage
PnR.cap_placer.Placer_Router_Cap.Placer_Router_Cap INFO : Begin CC Capacitor Placement and Router
PnR.cap_placer.Placer_Router_Cap.Placer_Router_Cap INFO : End CC Capacitor Placement and Router
Segmentation fault (core dumped)
```

Lastdayends, 2021-02-26:
@854768750 

Could you help to look into the ordering bug?

854768750, 2021-02-27:
[image]
@soneryaldiz the problem is here, the netlist uses "mn0_mp0" while the constraint uses a different name. I modified it manually (remove this ordering constraint) and there is not segment fault. @kkunal1408 Is there any method to solve this problem?
Here is the placement of top level.
[image]

closed 2021-03-24 by soneryaldiz